mavlink-router/mavlink-router #221: Many logging files after disarming

stephendade, 2020-06-16:
This occurs on ArduPilot and the ``--l`` option in mavlink-router

After arming and then disarming (and no more logging messages are sent), mavlink-router will keep timing out and restarting the logging component (creating new logging files), rather than shutting down logging. Output is below.

Perhaps logging should only create a new log file if/when logging packets are received, rather than first creating the logfile and then waiting for logging packets (and timing out).

```
stephen@latitude:~/Documents/UAV_code/mavlink-router$ ./mavlink-routerd -e 127.0.01:14550 -l ./binlog /dev/ttyACM0:115200
Could not open conf file '/etc/mavlink-router/main.conf' (No such file or directory)
Error while trying to write serial port latency: Unknown error -1
Open UART [4] /dev/ttyACM0 *
UART [4] speed = 115200
Open UDP [5] 127.0.01:14550  
Open TCP [6] 0.0.0.0:5760 *
Logging target system_id=1 on 00005-2020-06-16_18-04-43.bin
No Log messages received in 5 seconds restarting Log...
Logging target system_id=1 on 00006-2020-06-16_18-05-42.bin
No Log messages received in 5 seconds restarting Log...
Logging target system_id=1 on 00007-2020-06-16_18-05-53.bin
No Log messages received in 5 seconds restarting Log...
Logging target system_id=1 on 00008-2020-06-16_18-06-04.bin
No Log messages received in 5 seconds restarting Log...
Logging target system_id=1 on 00009-2020-06-16_18-06-15.bin
No Log messages received in 5 seconds restarting Log...
Logging target system_id=1 on 00010-2020-06-16_18-06-26.bin
No Log messages received in 5 seconds restarting Log...
Logging target system_id=1 on 00011-2020-06-16_18-06-37.bin
No Log messages received in 5 seconds restarting Log...
Logging target system_id=1 on 00012-2020-06-16_18-06-48.bin
```

stephendade, 2020-06-16:
I should clarify, this is with the LOG_BACKEND_TYPE parameter set to mavlink and LOG_DISARMED set to 0

highfreq, 2021-10-18:
Yes arming disarming just once and it creates a new file every 15 seconds.