mavlink-router/mavlink-router #223: Systemd not restarting mavlink-router after failure.

andriots, 2020-07-09:
Hello. I'm having trouble getting mavlink-router to autostart at boot.

When Nvidia Xavier XN boots up the service fails, it does not restart itself as it is defined in the unit file.

```
:~$ sudo systemctl status mavlink-router
● mavlink-router.service - MAVLink Router
   Loaded: loaded (/lib/systemd/system/mavlink-router.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Thu 2020-07-09 12:06:36 EEST; 51s ago
  Process: 4564 ExecStart=/usr/bin/mavlink-routerd (code=exited, status=0/SUCCESS)
 Main PID: 4564 (code=exited, status=0/SUCCESS)

juuli 09 12:06:23 apsync mavlink-routerd[4564]: Open UDP [9] 10.0.1.255:14550
juuli 09 12:06:23 apsync mavlink-routerd[4564]: Open UDP [10] 10.0.1.255:14668
juuli 09 12:06:23 apsync mavlink-routerd[4564]: Open UDP [11] 10.0.1.255:14669
juuli 09 12:06:23 apsync mavlink-routerd[4564]: Error while trying to write serial port latency: Unknown error -1
juuli 09 12:06:23 apsync mavlink-routerd[4564]: Open UART [12] /dev/ttyTHS0 *
juuli 09 12:06:23 apsync mavlink-routerd[4564]: UART [12] speed = 1500000
juuli 09 12:06:23 apsync mavlink-routerd[4564]: Open TCP [13] 0.0.0.0:5760 *
juuli 09 12:06:28 apsync mavlink-routerd[4564]: 5 messages to unknown endpoints in the last 5 seconds
juuli 09 12:06:33 apsync mavlink-routerd[4564]: 5 messages to unknown endpoints in the last 5 seconds
juuli 09 12:06:36 apsync mavlink-routerd[4564]: poll error for fd 12, closing it

```
From the code I can see a comment, that when the last line occurs then mavlink router is killed  so external (in this case systemd) component can restart it, however this is not the case. When I log in and start the service manually (systemctl start mavlink-router), everything starts working fine but this is not desired since I need everything to start running after power is plugged into the Xavier.

Lines that cause the last error:
mainloop.cpp lines 297-303:

```
if (events[i].events & EPOLLERR) {
                log_error("poll error for fd %i, closing it", p->fd);
                remove_fd(p->fd);
                // make poll errors fatal so that an external component can
                // restart mavlink-router
                request_exit();
            }
```

Systemd unit file(exactly the same as in this repository):
```
:~$ cat /lib/systemd/system/mavlink-router.service 
[Unit]
Description=MAVLink Router

[Service]
Type=simple
ExecStart=/usr/bin/mavlink-routerd
Restart=on-failure
[Install]
WantedBy=multi-user.target
```

lucasdemarchi, 2020-07-09:
> if (events[i].events & EPOLLERR) {
>                 log_error("poll error for fd %i, closing it", p->fd);
>                 remove_fd(p->fd);
>                 // make poll errors fatal so that an external component can
>                 // restart mavlink-router
>                 request_exit();

it should request_exit and force the exit status to be an error. Otherwise systemd won't restart it. See:

>   Process: 4564 ExecStart=/usr/bin/mavlink-routerd (code=exited, status=0/SUCCESS)

Also, would be great to check why it's giving you an epoll error to fix so it actually doesn't exist. At first I thought about the device node not existing yet. But if this happens it should give you an earlier error while trying to open the device.

andriots, 2020-07-10:
Thank you for your reply lucasdemarchi. I'm quite new to UAV technology but I am experienced in GNU/Linux. I'm looking into the epoll error and how to debug it.

Right now I confirmed that device node not existing is not the problem. I just booted up the Xavier and mavlink router started without problems. After that I powered up the UAV - same error occured on mavlink-router and I had to manually restart it. After that everything started working fine.

I tested with several baud rates and the problem persists. Could I provide some additional information about my configs? 

TELEM2 settings:
```
SERIAL2_PROTOCOL = 1 
SERIAL2_BAUD = 1500000
LOG_BACKEND_TYPE = 3
BRD_SER2_RTSCTS = 0
```

MaximeAubanel, 2020-09-17:
Same problem.

galdalali, 2021-02-18:
Hi,

I'm experiencing the reset issue as well. How could I restart the service properly?
@andriots, did you suffer from instability issues with mavlink-router as well? sometimes when starting mavlink-router I get pull error - [https://github.com/mavlink-router/mavlink-router/issues/244](url)

jbeyerstedt, 2021-04-29:
@andriots Could you please try to re-produce your issue with the latest master? Lucas recently merged a patch for a poll error on the TCP endpoint, maybe it has also fixed this issue. At least, it should exit with an error code now, so that systemd will restart the service.

lucasdemarchi, 2021-04-30:
I think it did fix. Our systemd.service has "Restart = on-failure", and we weren't exiting with an error code when a TCP connection dropped for example.

jbeyerstedt, 2021-04-30:
But this is for UART poll error and I wasn't sure, if that's fixed, too.

paravarzar-shahrokh, 2021-05-18:
Hi everyone, I am facing the same issue as well,  "Could not reset uart", What is the final solution for this?
Thank you

paravarzar-shahrokh, 2021-05-26:
> Sorry for the delayed answer, I'm currently physically located far from the hardware so I can't test whether this issue persists. I did find a workaround by editing the systemd unit file:
> 
> Restart = on-failure
> to
> Restart = always
> 
> I'll try to reproduce this bug with latest master once I get near the hardware.

Is this somewhere in the code that we can change?

jbeyerstedt, 2021-05-26:
Changing the systemd unit file is not solving the underlying issue, so I would like to keep the discussion in this issue related to the the actual cause of the poll error.
Actually there are two things to consider: What is causing the UART poll error in the first place and why isn't the poll error leading to a non-zero exit code (if the latter one wasn't already fixed by the latest changes to the master).

Of course you can modify the file [in the repo](https://github.com/mavlink-router/mavlink-router/blob/master/mavlink-router.service.in) and install again or just edit the installed file in /etc/systemd/system/mavlink-router.service for a quick fix. But I wouldn't approve this change as a fix, because we should get the exit code right instead.

lucasdemarchi, 2021-05-26:
> Is this somewhere in the code that we can change?

@paravarzar-shahrokh  did you try with recent master branch? As I said it should have been fixed already.

lucasdemarchi, 2021-05-26:
> But this is for UART poll error and I wasn't sure, if that's fixed, too.

if an UART endpoint sets EPOLLERR, it will end up logging this:

```
poll error for fd XX
Critical fd XX got error, exiting
```

And then call `request_exit(EXIT_FAILURE);`. the return code from main() will be EXIT_FAILURE.

paravarzar-shahrokh, 2021-05-26:
I tried the master brach but I got error

paravarzar-shahrokh, 2021-07-29:
On the first run using the command line and XAVIER I got this:
`rudi@rudi-desktop:/etc/mavlink-router$ sudo mavlink-routerd
Open TCP [4] 127.0.0.1:3001
Open UDP [5] 127.0.0.1:2222  
Error while trying to write serial port latency: Unknown error -1
Open UART [6] /dev/ttyTHS0 *
UART [6] speed = 57600
Open TCP [7] 0.0.0.0:5760 *
poll error for fd 6
Critical fd 6 got error, exiting
`
on the second run everything looks fine :
`
Open TCP [4] 127.0.0.1:3001
Open UDP [5] 127.0.0.1:2222  
Error while trying to write serial port latency: Unknown error -1
Open UART [6] /dev/ttyTHS0 *
UART [6] speed = 57600
Open TCP [7] 0.0.0.0:5760 
`
Any thoughts? I am using the latest version of mavlink-router
Regards

jbeyerstedt, 2021-09-09:
I'll close this issue for now, because it should be fixed. @paravarzar-shahrokh please refer to #244 for the UART port issue.

closed 2021-09-09 by jbeyerstedt