mavlink-router/mavlink-router #224: 2 Mavlink-router interference

TechNaturalist, 2020-07-23:
This is an odd use case that I’m hoping someone can help me with.

I have a drone using mavlink router to support intercomponent communication and I use mavlink router on the ground station to have multiple ground stations.

At about 2 minutes, the communication will cease to be 2 way. The messages are still being sent up to the drone but the drone no longer routes messages down.  

Can two mavlink routers work together in this configuration?

lucasdemarchi, 2020-07-23:
As long as they are running in different machines, should work.  How is the communication between GCS and drone, radio or IP-based?   If it's IP, you don't need mavlink-router in the GCS and can just add additional endpoints.

TechNaturalist, 2020-07-27:
It is radio based.

I appreciate the response.  After some more testing it seems that it might be the way mavlink-router and QgroundControl interact on Linux.  The problem, in the best way I can describe it is:  QGroundControl tries to take over the control of the radio port /dev/ttyUSBX and when it does there is a chance that mavlink-router will screw up the endpoints.

to be more specific, it will start to see the components on the plane (comp id 240 & 243) as targets for the radio.  When that happens that is when the communication from those sources ceases to the ground station. 

If I have one mavlink-router (as opposed to two) or I turn off the SiK radio autoconnect the problem then disappears.  Unfortunately, I work in a research lab and the pilots insists the Sik radio autoconnect or they will not fly (for safety reasons).

Would you have an idea as to why this behavior shows up?

lucasdemarchi, 2020-07-27:
If both QGC and mavlink-router try to control a single device, then there is no way it can work since they will each consume part of the messages.  You must assign that device to each QGC or mavlink-router, but not both. Sik can still "autoconnect", but if you are bridging the connection with mavlink-router, it's mavlink-router who should control it.

If you want QGC to be the first to receive, then you have to make QGC be the bridge. It may have an option to do that (I know mavproxy has the --out to allow connecting other GCSs).

jbeyerstedt, 2021-02-10:
It seems to me, that this question is answered. Please comment, if I should re-open it.

closed 2021-02-10 by jbeyerstedt