mavlink-router/mavlink-router #226: Stopping sending bin logs after battery failsafe

Dan-Moss, 2020-09-17:
Mavlink router is stopping sending bin logs over to a companion computer when a battery failsafe is happening.

lucasdemarchi, 2020-09-21:
Does mavlink-router stops it or is it ardupilot? And what failsafe?

Dan-Moss, 2020-09-23:
> Does mavlink-router stop it or is it ardupilot? And what failsafe?

mavlink router stops it / splits up the log. the bin log on the FCU is complete armed/disarmed
any failsafe i have seen, so battery/rc/terrain

Dan-Moss, 2020-10-28:
@lucasdemarchi the cause is that arducopter hits a failsafe, sets `MAV_STATE_CRITICAL`
mavlink router only looks for `MAV_STATE_ACTIVE` 

https://github.com/intel/mavlink-router/blob/2d095f315cf9a2a1c17c5d80657d2e42c6fb390c/src/mavlink-router/logendpoint.cpp#L439

https://github.com/ArduPilot/ardupilot/blob/04633d7e1c20a5ede7d1dafbd2e73f822acddaba/ArduCopter/GCS_Mavlink.cpp#L75

lucasdemarchi, 2020-10-28:
Nice. Could you send a PR with that fix?

Dan-Moss, 2020-10-30:
I see a couple of options:

using `MAV_STATE_ACTIVE` as a proxy for armed could also have critical and emergency added as to keep logging through a failsafe. however adding these would mean if a failsafe happened while disarmed this could create a log, so we could have a check to say only after it is active to then keep logging if critical and emergency. But in the case of a Battery low failsafe this could log until rebooting since the failsafe continues after landing. we could then look for flight termination as a way to stop the log?
[image]

Another option is to use `MAV_MODE` and check all the armed modes?
[image]

Do you have any thoughts?

Dan

lucasdemarchi, 2020-10-30:
I think it's safe to change it to log on `state >= MAV_STATE_ACTIVE`. We rotate the log nonetheless if there is a timeout without receiving log data. 

If using mav_mode we should probably do:

```c++
bool armed = base_mode & MAV_MODE_FLAG_SAFETY_ARMED
```

Not sure why we didn't use it in the first place since it seems to be the correct check.

lucasdemarchi, 2020-10-30:
@bkueng  this comes from your commit 1b48f691. AFAICS it would be ok for Ardupilot (test is needed). Would it work for PX4 as well?

bkueng, 2020-11-02:
Yes, and using the base_mode would indeed be better, not sure why I didn't do the in the first place.

PR #234 (2020-11-06): log: use base_mode rather than system_status
system_status is not a good field to get the "armed" state from: the
flight stack may change the system_status if for example a failsafe
condition is hit.  Derive the "armed" check from the base_mode field
that has a flag exactly with this meaning.

Thanks to Dan Moss for finding the issue.

Fix #226.
commit 51c72ffd (2020-11-06): log: use base_mode rather than system_status

system_status is not a good field to get the "armed" state from: the
flight stack may change the system_status if for example a failsafe
condition is hit.  Derive the "armed" check from the base_mode field
that has a flag exactly with this meaning.

Thanks to Dan Moss for finding the issue.

Fix #226.
closed 2020-11-06 by bkueng