mavlink-router/mavlink-router #228: Is it possible route UDP to Serial

vinura, 2020-10-02:
I am using ESP8266 AP station on the drone to receive telemetry data.

I want to receive telemetry data on rPi and route it to a serial device so I can plug in this http://tengriuav.com/ unit. So it can communicate with the fc.

I tried using

`mavlink-routerd -e /dev/ttyS0:921600 192.168.4.1:14550`

but ended up with the error

Could not open conf file '/etc/mavlink-router/main.conf' (No such file or directory)
Error binding socket (Cannot assign requested address)
Could not open 192.168.4.1:14550

Key Info

ESP8266 is the AP station. It output the mavlink2 data to 192.168.4.1:14550
I can receive it directly to mission planner/gcs without any problem

I failed to route the UDP endpoint back to serial. Any help is welcome

jbeyerstedt, 2020-11-12:
Forwarding from UDP to serial works just fine at our setups.

The error having no config file should be no issue even though I think using the config file is easier to read than the command line arguments. You have to use an UDP endpoint in Eavesdropping mode then. That mode will listen for incoming UDP messages on a defined port.

The second message looks like you tried to bind to an IP which is not assigned to any of the network interfaces. Have you checked, that the device running mavlink-router has the IP 192.168.4.1? But you could also just listen on all interfaces with 0.0.0.0 when the device does not have a static IP.

jbeyerstedt, 2021-01-27:
I'll close this because of inactivity for now. Just comment, if you have any updates.

closed 2021-01-27 by jbeyerstedt