mavlink-router/mavlink-router #229: Runing with SITL

galdalali, 2020-10-13:
Hey,

I'm running mavlink-router with ArduPilot SITL using the next configuration: 

```
[General]
TcpServerPort=0
ReportStats=false
MavlinkDialect=auto

[TcpEndpoint SITL]
Mode = Normal
Address = 127.0.0.1
Port = 5760
RetryTimeout = 10

[UdpEndpoint VideoCore]
Mode = Normal
Address = 127.0.0.1
Port = 14553

[UdpEndpoint GCS]
Mode = Normal
Address = 192.168.1.203
Port = 14551
```

I can see that SITL sends only heartbeat and timesync messages (to VideoCore UartEndpoint).
Does ArduPilot waiting for something? or is it some limitation of mavlink-router?
Using MavProxy instead of mavlink-router passes all messages.
Also, when I'm connecting  Mission Planner (GCS UdpEndpoint) to SITL VIA mavlink-router then it start sending all the messages but I cant figure out if Mission Planner does something special when connecting: [https://github.com/ArduPilot/MissionPlanner/blob/e7337db1cbce492132a45eb64a3dd7b8ccc801ab/MainV2.cs#L1458](url)

What am I missing?

lucasdemarchi, 2020-10-13:
mavproxy is not only a bridge/router, but also a GCS.  There are 2 points in which ArduPilot will wait. By default it will wait for a TCP connection on port 5760. If you start only sitl you will see:

```
$ ./build/sitl/bin/arducopter -w --model 
Starting sketch 'ArduCopter'
Starting SITL input
Using Irlock at port : 9005
bind port 5760 for 0
Serial port 0 on TCP port 5760
Waiting for connection ....
```

You can bypass that wait by specifying the argument `--uartA tcp:5760`.

After that, ardupilot will basically only send heartbeat until it listens to a heartbeat from another system. I guess your VideoCore endpoint is not sending a heartbeat, right? When you start the GCS this is "fixed" because the GCS sends the messages to Ardupilot. In the case of mavproxy, the GCS is embedded in the same tool.

galdalali, 2020-10-14:
I honestly don't know what am I doing wrong.. I'm sending the heartbeat (and a TymeSync msg) with the same sysid as the ArduPilot FMU (1) and a different component id (2) just after I get an heartbeat from the FMU, and I'm pretty sure I'm sending it right..
I even checked in the mission planner Mavlink inspector - here's the [link](https://ibb.co/F59r5pq)

Maybe I'm missing something out?