mavlink-router/mavlink-router #230: Question: - mavlink-router to mavlink-router-2

ACPAviation, 2020-10-18:
Hi there,
I have installed mavlink-router. However, how do I install mavlink-router-2?
I see the scores code is here: - [https://github.com/intel/mavlink-router/releases/tag/v2](url)
Is this now in master, I.e. cloning https://github.com/intel/mavlink-router.git, is this the code for V2

If not, how do I install V2?

thanks

ACPAviation, 2020-10-18:
Ubuntu 18.04.5 LTS Kernal 5.4.0-51

Could not reset uart
mavlink-routerd -e 192.168.7.1:14550 -e 127.0.0.1:14550 /dev/ttyACM0:115200

ACPAviation, 2020-10-18:
Error while trying to write serial port latency: Unknown error -1
Open UART [4] /dev/ttyACM0 *
UART [4] speed = 115200
Open UDP [5] 127.0.0.1:14550
Open UDP [6] 192.168.7.1:14550
Open UDP [7] 127.0.0.1:14655
Open UDP [8] 127.0.0.1:14551
Open UDP [9] 127.0.0.1:14556 *
Open UDP [10] 127.0.0.1:14560
Open UDP [11] 127.0.0.1:14755
Open UDP [12] 10.0.1.255:14550
Open UDP [13] 10.0.1.255:14668
Open UDP [14] 10.0.1.255:14669
Could not reset uart

jbeyerstedt, 2021-02-10:
MAVLink-router v2 is one point in time of the master branch. So cloning the master is perfectly fine.

The issue with UART looks like a duplicate of #213 (you already commented there), so I would like to close this, when you don't have any objections.

closed 2021-02-10 by jbeyerstedt