mavlink-router/mavlink-router #232: Corrupted mavlink-router data

dhauptman, 2020-10-29:
Hello,

I was running a simulation that involved four agents and QGC and noticed that at some point the simulated aircraft on the ground station disconnected. Mavlink-router connects the serial link to the simulation on each aircraft with the command `mavlink-routerd /dev/ttyUSB0:230400 -e 127.0.0.1:3000`, where QGC is just connected to the serial link. Each aircraft is sending a custom message at a rate of 5 Hz with a 1 Hz ping over pymavlink. The four aircraft system id's are just 1 to 4 with QGC set to 255.

Running mavlink-router with the command `mavlink-routerd /dev/ttyUSB0:230400 -e 127.0.0.1:3000 -v` on one of the aircraft showed the output below.

It continued to output those values even after shutting down 3 simulations and only having one aircraft running. Scoping out the data on the simulations showed that values being sent to pymavlink were expected values.

I'm new to mavlink-router so I'm not sure if it is receiving corrupted data or if somehow it has desynched from the mavlink packets and is always starting the search in the middle of a packet.

Verbose Output: 
Endpoint [5] got message 16645629 to -1/-1 from 253/253
	Known components:
		1/1
Endpoint [5] accepted message 16645629 to -1/-1 from 253/253
UDP [5] wrote 265 bytes
UART [4] got 25 bytes
UART [4] got 23 bytes
UART [4] got 23 bytes
UART [4] got 91 bytes
UART [4] got 25 bytes
UART [4] got 15 bytes
UART [4] got 31 bytes
UART [4] got 23 bytes
UART [4] got 23 bytes
No message entry for 16645558
Endpoint [4] got message 16645558 to -1/-1 from 253/253
	Known components:
		255/0
		2/1
		3/1
		4/1
		1/1
Endpoint [4] accepted message 16645558 to -1/-1 from 253/253
UART [4] wrote 278 bytes
Endpoint [5] got message 16645558 to -1/-1 from 253/253
	Known components:
		1/1
Endpoint [5] accepted message 16645558 to -1/-1 from 253/253
UDP [5] wrote 278 bytes
UART [4] got 25 bytes
UART [4] got 21 bytes
UART [4] got 25 bytes
UART [4] got 21 bytes
UART [4] got 25 bytes
UART [4] got 21 bytes
UART [4] got 23 bytes
UART [4] got 18 bytes

dhauptman, 2020-11-04:
I think I got it sorted out. It seemed that the message I am sending fragmented. Mavlink-router interpreted this as a unknown but valid message, as the target system id was assigned to -1. Mavlink-router would then try and route the corrupted data. It seemed that mavlink-router would route the data faster than I had been sending it which started to eat up the wireless resources and caused the other units to fail. 

I made a simple test patch on my system by editing the handle_read function in endpoint.cpp to be:

```C
while ((r = read_msg(&buf, &target_sysid, &target_compid, &src_sysid, &src_compid, &msg_id)) > 0)
        if (r == 1) {
                Mainloop::get_instance().route_msg(&buf, target_sysid, target_compid, src_sysid,
                                                   src_compid, msg_id);
        }
```
By only routing known messages, it seemed to have fixed the issue. I know it goes against the design, but it could be useful as an option to only route known messages.

lucasdemarchi, 2020-11-04:
I'm not sure fragmentation is the right reason. The mavlink header contains the size of the payload. See Endpoint:read_msg(). If are not able to read the entire payload we will just buffer what we read and return 0.

IMO there is a flaw in mavlink protocol: we need to know the msg id to be able to check the crc and flags. A router/proxy may not be aware of all the messages if there are custom messages. If mavlink-router doesn't know about your custom message, then it has no way for it to check that.  It still checks the size, but it can't verify if there was corruption in the link.  If you are using a custom message, make sure mavlink_get_msg_entry() knows about it.

Also corruption is ultimately avoided by the **endpoints** checking the CRC.

dhauptman, 2020-11-04:
Fair enough, I only mentioned fragmentation when looking at the buffer and it seemed to have multiple partial custom messages within it. I also noticed that it was interpreting the buffer to be a mavlink 2 message when I have only been sending mavlink 1 messages.

I'll double check to make sure that mavlink_get_msg_entry() sees the message.

Thanks for the help!