mavlink-router/mavlink-router #233: autogen.sh fails in docker container for Raspberrypi 4

Crowdedlight, 2020-10-30:
When trying to build this V2 release in a docker container to run on a RPI4 running with balena as host-os I keep getting an error trying to run ``./autogen.sh``

I get the following error:
```
[Build]   main Step 14/21 : RUN ./autogen.sh
[Build]   main  ---> Running in 1cb888a0455f
[Build]   main aclocal: error: echo failed with exit status: 1
[Build]   main autoreconf: aclocal failed with exit status: 1
[Build]   Built 1 service in 0:05
[Error]   Build failed.
The command '/bin/sh -c ./autogen.sh' returned a non-zero code: 1
``` 

I do have the ``automake libtool`` installed.
I have also tried with ``autoconf`` installed and by running ``aclocal`` and ``autoconf`` before running ``authgen.sh`` which didn't work. 

I can't figure out what this error actually means or how to solve it. I have built and installed mavlink-router on our workstation vms running 18.04 and 20.04 with no issue. Any idea what the error is and how to get it working on RPI4?

My docker file is:
```
FROM balenalib/raspberrypi4-64-ubuntu:bionic-build

RUN [ "cross-build-start" ]

# possible fix
ARG DEBIAN_FRONTEND=noninteractive

# set timezone data
ENV TZ=Europe/Copenhagen
RUN ln -snf /usr/share/zoneinfo/$TZ /etc/localtime && echo $TZ > /etc/timezone

# Update and install
RUN apt-get upgrade -y
RUN install_packages git build-essential socat gcc g++ cmake python-future pkg-config python3-future python3 python3-pip python3-setuptools libtool automake

# Setup application folder
ENV APP=/workspace

# Enable UDEV for balena
ENV UDEV=on

WORKDIR $APP

# get release
RUN curl -LJO https://github.com/intel/mavlink-router/archive/v2.tar.gz && tar -xzf mavlink-router-2.tar.gz
WORKDIR /workspace/mavlink-router-2

# build it
RUN ./autogen.sh
RUN ./configure CFLAGS='-g -O2' --sysconfdir=/etc --localstatedir=/var --libdir=/usr/lib64 --prefix=/usr --disable-systemd
RUN make
RUN sudo make install

# Run entrypoint script
COPY entrypoint.sh $APP/
RUN chmod +x $APP/entrypoint.sh

CMD ["/workspace/entrypoint.sh"]

RUN [ "cross-build-end" ]
```

lucasdemarchi, 2020-11-03:
are you running on rpi4 or doing a cross-build?

Crowdedlight, 2020-11-03:
That was a very good question. I have been doing it cross-build sofar, without thinking about that making a difference. Just did a build on native RPI4, and it worked. I did however have to change to cloning the repo instead of downloading the release source files. Otherwise I would get the same error as here: https://github.com/intel/mavlink-router/issues/156 

But cloning and recursive getting the submodules worked.

Any ideas to why cross-build does not work? So far my pipeline has been setup for cross-build. I can change it, but would be interested in why it wouldn't work.

jbeyerstedt, 2021-07-28:
Sorry for asking again: Did the cross build work when doing a git clone with all submodules instead of downloading the release ZIP?

Because as you have seen, the release files which were created by github, unfortunately do not include the submodules.

Crowdedlight, 2021-08-01:
Yes, a clone with submodules worked as I recall. Been some time since I looked at it due to moving to other priorities. The release files missing the submodules could very well have been the issue.

jbeyerstedt, 2021-09-09:
So, I think we found the issue then. Git submodules are a cool feature, but GitHub (and others as well) seem to lack a more user-friendly zip-download. Solution: Use `git clone --recursive` or download the ZIP/ tar.gz files of the submodules as well.

Independent of this issue, I would recommend using the master branch instead of the tagged release, because we don't have a real release cycle yet and quite some things got patched in the meantime.

closed 2021-09-09 by jbeyerstedt