mavlink-router/mavlink-router #236: Windows Installation?

hashimK, 2020-11-11:
I tried installing mavlink-router on Windows by implementing linux equivalent commands for Windows but I was unable to do so. Hence, how to install mavlink-router in a Windows PC?
Does mavlink-router support Windows? If not, then I would like to know what is mavlink-router equivalent for Windows.
Thanks.

lucasdemarchi, 2020-11-11:
It probably works on WSL, but not sure anyone tested that

hashimK, 2020-11-11:
> It probably works on WSL, but not sure anyone tested that

I plan to use QGC and MAVSDK on the same computer. It works great on Ubuntu using mavlink-router. But I'm looking to make it work on Windows. Hence, looking for windows install instructions as I read (in mavlink-router vs mavproxy document) that mavlink-router is crossplatform.
WSL wouldn't help me for what I intend to do.
Any install instructions or some ideas would really help.
Thanks

lucasdemarchi, 2020-11-11:
It doesn't work on windows. mavlink-router uses linux-only features, so you will probably need to use other project, develop support for windows, use WSL, or use Linux.

hashimK, 2020-11-12:
Ok thanks for the information.

closed 2020-11-12 by hashimK