mavlink-router/mavlink-router #237: Upstreaming features

jkflying, 2020-11-11:
I'd like to try upstreaming the majority of the diff from the Auterion fork, is this something there would be interest/time in reviewing? Right now we are starting to diverge a bit, and particularly for things like CI I believe there would be considerable benefit for upstream. There's also been a refactor of pulling the main loop out into a separate function, allowing much easier unit testing.

Diff is here, please ignore the debian packaging since I don't think that makes sense for upstream given the rapidly changing mavlink spec:
https://github.com/intel/mavlink-router/compare/master...Auterion:auterion-router-latest

lucasdemarchi, 2020-11-11:
Yeah, that would be great.

For mainloop, there is also https://github.com/intel/mavlink-router/pull/143 that I never finished, which untangles mavlink from the mainloop itself. I'm also thinking about start using a library for that (ell and libuv being the current candidates).

lucasdemarchi, 2020-11-11:
@jkflying it would be great if in future you don't diverge so much. Get more people to agree on a feature and use this repo for discussion. Main requirement I'd think is not to break current users and avoid things that target a single flight stack.

jkflying, 2020-11-11:
Sounds good. The divergences I'm worried about are more in the line of refactoring, AFAIK there are no flight-stack specific features or serious backwards incompatibilities.

Going forward, could you (internally) bring up getting the project into something under the Dronecode / MAVLink / Linux Foundation umbrella rather than under Intel? mavlink-router has become a pretty central part of the drone ecosystem, and with success comes some difficulties :slightly_smiling_face: From our side, and given our company mission, it is also harder to justify upstreaming to something with corporate rather than community governance, as a strategic position. I believe there was some discussion about this before...

lucasdemarchi, 2020-11-12:
@jkflying yes, there were some discussions before (#164). It doesn't make much sense to me that you have to move a project under another umbrella to justify using or contributing to it.   With that argument you'd have to move several other projects like nuttx, linux kernel, systemd (or busybox, or whatever you are using in the companion computer), etc.  There isn't any type of corporate governance in this repo/project.

That said, and since I myself don't contribute to mavlink-router as my work for Intel anymore, I'd be willing to move it elsewhere that makes sense and gives the project a new home to grow. Considering the support for more than the dronecode/PX4 flight stack, options would be IMO github.com/mavlink-router/  or github.com/mavlink/. Thoughts?

jkflying, 2020-11-12:
IMO github.com/mavlink/mavlink-router would be the perfect place for mavlink-router to live. I'm not sure of the exact process and leave it in the capable hands of @hamishwillee / @mrpollo , I'll get back to the upstreaming of features :wink:

hamishwillee, 2020-11-12:
Not sure how a repo moves - is that  done via a PR or some other sort of handover. The process would probably be just to mention it on the devcall or a PR as a proposal. Then we get general agreement we want it (we probably do) and then it would be accepted.

lucasdemarchi, 2020-11-12:
@hamishwillee I have asked to the admins here and I have to fill out some forms. Not sure how the move is handled yet, but I guess the mavlink org in github will be contacted to receive it.

auturgy, 2020-11-13:
I agree with Lucas that moving this really shouldn’t be necessary.

tridge, 2020-11-13:
given the history of what Lorenz did with control of mavlink I would strongly prefer we not move more projects under that umbrella. Lorenz is still an admin of mavlink repo, and has a proven history of being a bad actor with inter-project cooperation

magicrub, 2020-11-13:
Moving from one company to another makes no sense at all. Moving it into Mavlink is the only real option worth considering.

lucasdemarchi, 2020-11-13:
I really didn't want to trigger this fight again. I guess the best course of action is to have a separate github.com/mavlink-router that everybody can collaborate to.

proficnc, 2020-11-13:
This must be independent, this has not been raised on the bi-weekly mavlink calls, and really should be.

hamishwillee, 2020-11-13:
IMO this should go into MAVLink. But it doesn't really matter as long as it has an active team of people who are maintaining it and able to review PRs etc.

hamishwillee, 2020-11-13:
Or to put it another way, Auterion are doing the most work with this at the moment. A public Auterion repo would make sense on that basis. However since this is a MAVLink tool for the MAVLink community, let's put it where it makes sense.

auturgy, 2020-11-13:
Let’s get mavlink governance resolved before adding additional complications

lucasdemarchi, 2020-11-14:
So, this is now out of github.com/intel and in github.com/mavlink-router. Due to the discussion above, a neutral place may be better.

lucasdemarchi, 2020-11-14:
I also went through the recently closed PRs and gave write access. If you think I've missed someone, let me know.

hamishwillee, 2020-11-17:
Thanks you @lucasdemarchi ! 

FYI, who are the "core team" maintaining this now? Might be good to add your people with merge rights to the README.

lucasdemarchi, 2020-11-17:
admin: I and @bkueng 
write perm:
@jbeyerstedt 
@jkflying 
@JonasVautherin 
@MatejFranceskin 
@olliw42 
@RomanBapst

closed 2021-02-02 by jkflying