mavlink-router/mavlink-router #239: Wrong logging target system_id

andreastoellefsen, 2020-11-19:
I'm connecting two groundstations on udp end points to one pixhawk running ardupilot. sys_id = 9. When logging on the computer connected to the pixhawk, i often tries to log from the wrong sysid. Most often it tries to log from sys_id = 252. Somtimes from sys_id = 4294967295
This seems to happen only when both GCS's are running.

correct output in terminal and working log:
`pilot@RoamerIndia:~$ mavlink-routerd -d mavlink-router/config.d/
Could not open conf file '/etc/mavlink-router/main.conf' (No such file or directory)
Open UDP [4] 10.19.127.1:15001  
Open UDP [5] 10.200.1.100:15000  
Open UDP [6] 127.0.0.1:14999  
Open UART [7] /dev/ttyUSB4 *
UART [7] speed = 921600
Open TCP [8] 0.0.0.0:5760 *
Logging target system_id=9 on 00047-2018-01-28_18-32-29.bin`

wrong output in terminal and not working log:
`pilot@RoamerIndia:~$ mavlink-routerd -d mavlink-router/config.d/
Could not open conf file '/etc/mavlink-router/main.conf' (No such file or directory)
Open UDP [4] 10.19.127.1:15001  
Open UDP [5] 10.200.1.100:15000  
Open UDP [6] 127.0.0.1:14999  
Open UART [7] /dev/ttyUSB4 *
UART [7] speed = 921600
Open TCP [8] 0.0.0.0:5760 *
Logging target system_id=252 on 00048-2018-01-28_18-32-41.bin`

crazy output in terminal and not working:
`pilot@RoamerIndia:~$ mavlink-routerd -d mavlink-router/config.d/
Could not open conf file '/etc/mavlink-router/main.conf' (No such file or directory)
Open UDP [4] 10.19.127.1:15001  
Open UDP [5] 10.200.1.100:15000  
Open UDP [6] 127.0.0.1:14999  
Open UART [7] /dev/ttyUSB4 *
UART [7] speed = 921600
Open TCP [8] 0.0.0.0:5760 *
lseek failed (Bad file descriptor)
Restarting log...
BinLog not started
Logging target system_id=4294967295 on 00064-2018-01-28_18-41-32.bin`

We're running a fairly complicated communications setup and this might be a corner case, but I kind of fell in love with the simplicity, the logging and the config compared to mavproxy. not to mention the gain in performance. Hope you can help

Andreas

lucasdemarchi, 2020-11-19:
What's the sysid of the GCS?  Have you tried removing one of them and checking if the problem goes away?

```
crazy output in terminal and not working:
pilot@RoamerIndia:~$ mavlink-routerd -d mavlink-router/config.d/ Could not open conf file '/etc/mavlink-router/main.conf' (No such file or directory) Open UDP [4] 10.19.127.1:15001 Open UDP [5] 10.200.1.100:15000 Open UDP [6] 127.0.0.1:14999 Open UART [7] /dev/ttyUSB4 * UART [7] speed = 921600 Open TCP [8] 0.0.0.0:5760 * lseek failed (Bad file descriptor) Restarting log... BinLog not started Logging target system_id=4294967295 on 00064-2018-01-28_18-41-32.bin
```

Here it seems there is something wrong with the restart logic. "lseek failed (Bad file descriptor)" happens when we are trying to write to a closed file.  That system_id is `(uint16_t) -1`.

Another possible reasons... not sure if that's what happens when the fs is unmounted with an open fd . Are you logging to a removable media like usb disk or sd card?  Can you check if that media is being removed when that happens (check dmesg)? 
it seems you are receiving

lucasdemarchi, 2020-11-19:
another check for you: the target system id is set to the first autopilot we receive a heardbeat from. I.e., please check if your GCS is not sending a heartbeat with component_id == MAV_COMP_ID_AUTOPILOT1:

```
MAV_COMP_ID_AUTOPILOT1=1, /* System flight controller component ("autopilot"). Only one autopilot is expected in a particular system. | */
```

andreastoellefsen, 2020-11-19:
> What's the sysid of the GCS? Have you tried removing one of them and checking if the problem goes away?
>
> ```
> crazy output in terminal and not working:
> pilot@RoamerIndia:~$ mavlink-routerd -d [...]

I have not checked the sys_id of the GCS but have suspected that it is 252 and that has tried to "grab" the logging function. 

The log is written to the emmc of the computer running the OS. I'm stopping and starting to deliberatly trigging the behavior, so it might not be fair in most cuircumstances, however when we in the next step automate the starting and stopping of the mavlink-routerd, we will have to implement a timeout which might be undesirable.

andreastoellefsen, 2020-11-19:
> another check for you: the target system id is set to the first autopilot we receive a heardbeat from. I.e., please check if your GCS is not sending a heartbeat with component_id == [...]

might I suggest adding a key in the logging routine with a sys_id to look for, default being first come first served?