mavlink-router/mavlink-router #240: Add message de-duplication/ redundant data links

jbeyerstedt, 2020-11-23:
Some days ago there was a small discussion about the support of redundant telemetry links to the FMU, so let's move the requirements here.
From my understanding, there are two similar, but different use cases, which a de-duplication feature should solve.

## Redundant Data Links
In some use cases it might be useful to have a redundant telemetry link between UAS and ground station. Both connections might be handled by a companion computer running mavlink-router on the UAS and another mavlink-router instance on the ground.
The current implementation will either lead to message loops or systems being falsely detected on an endpoint, because all incoming messages are routed out on all endpoints. And even when both links are properly used simultaneously, the FMU and ground station will still receive messages multiple times.

A solution for de-duplication might be to monitor the link quality (HEARTBEAT timeouts, estimate drop rate, etc) any only pass on messages of one of the redundant links. This would mean, that the config has to specify the "type" of the endpoint more precisely (e.g. grouping them to one redundant link and/or specifying communication directions).

## Mesh Network
Another use case is a mesh network for UTM and telemetry purposes. This will lead to even more advanced de-duplication requirements, because multiple "remote" mavlink-router instances will be received on one wireless link. The goal would be to have a "wave front" of each message propagating through the network, but no duplicated messages behind the wave front.

The amateur radio community already has some solutions for this issue. One of them is to cache the message counter (which is included in each MAVLink message) of each message for some time. This cache would be needed for each endpoint and a message is only transmitted through that endpoint, if it wasn't already transmitted once (the message counter isn't higher, than the cached value).

christophkTEA, 2020-12-16:
The mavlink message signing mechanism could be used for avoiding duplicate messages (https://mavlink.io/en/guide/message_signing.html). You can already tell, by checking the message signature, if you get the same message over the same link twice. If that is the case the router just drops the new message. If the router compares the mavlink messages to those previously received messages and they are the same except for the link id and signature field, then the router has received the same message before over a different link and can discard the new message.
This would also add to the security of such a network because each message is signed with the private key of the sender.

jbeyerstedt, 2021-01-27:
That's a good idea, I just have to check, if message signing is already activated in PX4 or how to activate it. (I've already seen, that there is a MAVLink message to setup the secret key).
But the issue with using the signature is, that not everybody populates the value (as far as I know). So we might need a fallback.

What I would do first in the next weeks is to implement an option to group endpoints together, e.g. for redundant data links. The feature of a group would be, that messages received by any endpoint in a group get forwarded to endpoints outside the group, but not to members of the same group. Messages coming into the group will be presented to all endpoints. Architecture-wise a group would act as an endpoint to the current "routing core" of mavlink-router and therefore "hides" the endpoints belonging to it. This should work more or less out-of-the-box when based on #235, because that PR will not send messages back to the endpoint, where the message was received from in all cases.
The grouping-feature would already mitigate some issues with message routing loops, but maybe we should discuss this in another issue.

jkflying, 2021-02-01:
Perhaps another way would be to have a FIFO of hashes of recent messages, and perform some deduplication over a time period? This can be done efficiently via using a deque to do the FIFO plus an unordered_map which is kept in sync with the FIFO to do the deduplication lookups. 

Even if duplicate payloads are sent (I'm thinking manual control etc), the timestamps should still be unique --> unique hash, so this should be a reasonable approach. We'd have to re-hash the whole message though, because the extension fields aren't covered in the normal mavlink hashes.

jkflying, 2021-02-02:
I've put up a sample implementation of the actual deduplication code:
https://github.com/Auterion/mavlink-router/pull/36/files

It would still need to be inserted into the message handling, and be able to be configured from the config file.

jkflying, 2021-02-03:
Ok, I've now inserted the deduplication into the message handling. The only part that is missing is being able to set the timeout from the config file (by default it is 0, and just short-circuits everything so it doesn't add any overhead).

lucasdemarchi, 2021-02-03:
> because all incoming messages are routed out on all endpoints.

that is a false premise. Messages to a certain target system/comp are not sent to an Endpoint if that system/comp was seen as source there.  This was done exactly to avoid loops.

jbeyerstedt, 2021-02-03:
> Messages to a certain target system/comp are not sent to an Endpoint if that system/comp was seen as source there. This was done exactly to avoid loops.

Are you sure, that it's this way around? IIRC, a message with the same _source_ (not target) system/comp ID is rejected by an endpoint, which already has received data from this source system/comp ID. Your case should (and will) accept the message. (See [src/mavlink-router/endpoint.cpp line 343 ff.](https://github.com/mavlink-router/mavlink-router/blob/master/src/mavlink-router/endpoint.cpp#L343))

But you're right, I was a bit too sloppy in my wording. I know, that there is filtering in place for multiple purposes. But as discussed in #235 we need bi-directional MAVLink broadcast for UTM communication, which definitely lead to message loops when having two redundant modems on one side.

lucasdemarchi, 2021-02-04:
I'd say the proposal is mixing terms here. "Redundant data link" is what we already have, and doesn't fit all needs. The proposal is more like a "Failover data link", i.e. there is one additional link that is only activated once the other fails.

I'm still unsure message de-dup is the right thing to do actually. Are we sure this would work, for example, when we have flight stack log over mavlink and the flight stack has to re-transmit due to packet loss?

jkflying, 2021-02-04:
> Are we sure this would work, for example, when we have flight stack log over mavlink and the flight stack has to re-transmit due to packet loss?

That's a good observation. There are several ways I see working: 

1. Add logic to allow duplicate messages if they are from the same source
2. Enable deduplication selectively on certain links only
3. Wait for the dedup buffer to time out, so that the retransmit will be allowed through (ie. what will happen if we don't handle this)
4. Make sure that retransmits at least have some kind of unique field, ie. timestamp of sending or something. This might already be the case but would need to be verified. 
 
This whole problem is very annoying, because we are having to mix transport logic with application logic... having the retransmits happening at the application layer instead of the transport layer makes this far harder than it should be.

dk7xe, 2021-02-04:
My 2 cents .. or a bit more 

What we like to achieve is a network infrastructure as shown in the picture
[image]
We have several ground station receivers that are forming a network and sending the MavLink Data to a central Hub that acts as a Gateway to USSP services. The Hub is a MavLink router instance that will get the same messages multiple times in case of a drone is in radio range of multiple receivers. But we like to send a message just once to the USSP service.
To make things worse (from a message duplication point ov view) each ground station receiver is equipped with 2 RF modems with antenna diversity to have 4 high gain antennas covering 360° horizontal.
This 2 RF modems deliver the over the air received MavLink data stream via 2 independent UDP ports to a local mavlink-router instance. This local mavlink-router instance connects via an third UDP port to the central Hub.
So we have multiple scenarios where duplication can happen.
a) if a drone is close to a ground station and both RF modems receive the signal (de-duplication need at ground station).
b) if a drone signal strenght is high enough to be received by multiple ground stations (de-duplication need at central Hub).

But still we need to make sure that UTM service messages are broadcastet via all groundstation only once. And not ping-pong played between the two RF modems at a ground station (as it happens currently).

jbeyerstedt, 2021-02-10:
I had a few more thoughts about our setup and the needs when running redundant and fail-over data links. A part of my message loop issue was definitely introduced by my proposed "multicast" UDP endpoint mode, but that's fixed now leading to a stalled communication instead.

Let's assume having two independent data links between the UAV (with a companion computer) and a ground station:
- mavlink-router on the UAV: One UART endpoint connected to the FMU, two UDP endpoints (normal mode) for the two data links
- mavlink-router on the ground: two UDP endpoints (eavesdropping mode) for the data links, one UDP endpoint (normal) to QGroundControl (or any other software)

This will result in this message sequence (correct me, if I'm wrong):
1. The FMU will start sending heartbeats, which will be transmitted over both data links by the air mavlink-router.
2. The ground mavlink-router will receive the heartbeat on one UDP endpoint (e.g. link1) first and forwards it to QGC. The second "incoming" UDP endpoint can't send now, because no client connected yet.
3. The ground mavlink-router will receive the heartbeat on the other "incoming" UDP endpoint (link2) and forwards to QGC. The first "incoming" endpoint will reject the message because it already has registered the source system in the list of connected systems.
4. QGC will send with a heartbeat, which the ground mavlink-router forwards over both data links
5. The air-side mavlink-router receives the heartbeat on one UDP endpoint (e.g. link1) first, forwards to the FMU **and** the second UDP endpoint (link2). (_This is where the trouble starts_)
6. The air-side mavlink-router receives the heartbeat on the second UDP endpoint (link2), forwards to the FMU while the first UDP endpoint (link1) will reject because the source system is already known in the list of connected systems.
7. The ground mavlink-router will receive the QGC heartbeat, which was sent back by the air mavlink-router in step 5, over link2 and transmits again on the link1 data link while the UDP link to QGC will reject (because of the list of connected systems).  
(_The link2 data link will not transmit QGC heartbeats again, because QGC is in the list of connected systems._
8. The air-side mavlink-router receives the QGC heartbeat for the second time over link2, forwards to the FMU (again) while link1 rejects the message, because the source system is already known in the list of connected systems.

In the end, the link2 endpoint on the ground will have saved QGC's system and component ID as connected, so no **more data from QGC will be sent over that link**. The other connected systems lists are correct.
The reason for this is, that the first message of each system received by the UAV's mavlink-router will be looped back on the other data link. Another issue might be, that the UAV will receive this message twice.

As far as I can tell now, there are two possible solutions:
- Grouped endpoints on the air-side mavlink-router, which prevent a message received on one endpoint of the group to be sent via another endpoint of that group. (Tackle the issue at it's root.)
- Message de-duplication on the ground-side mavlink-router, which has to discard the looped back (duplicated) message **before** the system is added to the list of connected systems on that endpoint. (Easier to implement, than changing the whole routing structure (and config parsing)?)

P.S.: The system of Gerald and me, which uses two modems on the same frequency at the ground station, introduces a "direct" message loop between the two modems. This leads to even more trouble, which shouldn't be the priority now.

lucasdemarchi, 2021-03-11:
@jbeyerstedt yes, you described well the problem now and it's due mavlink-router trying to discover the "network topology" in runtime and that not really working well when you have 2 data links connecting to the same thing.

I think extending the config to add static routing and/or grouping links would be doable. The group is something that should be very easy to do and interfere very little with the rest of the logic.

For message deduplication as we were discussing before is not so easy and there are potential problems due to re-transmissions. I'd rather focus on solving the problem you have. Btw, by solving that you could later extend it to allow the links in a group to act in a failsafe manner, so the second link is not always alive.

jbeyerstedt, 2022-02-17:
Since PR #330 was replace by #338 during my vacation, all of the required component for to close this issue are merged into master. Thanks to all participants.

closed 2022-02-17 by jbeyerstedt