mavlink-router/mavlink-router #241: Route data from 2 serial ports (2 drones connected to same computer)

hashimKAifi, 2020-11-25:
I'm facing an interesting challenge of reading telem data of 2 different drones in MAVSDK. 
The aim is to find distance between the 2 drones using telemetry data. I'm using MAVSDK-python to get drone data and mavlink-router to read the data from 2 serial ports.

My main.conf file looks like:
```
[General]

[UartEndpoint sik_drone1]
Device = /dev/ttyUSB0
Baud = 57600

[UdpEndpoint sik_drone1]
Mode = Normal
Address = 127.0.0.1
Port = 11000

[UartEndpoint sik_drone2]
Device = /dev/ttyACM0
Baud = 57600

[UdpEndpoint sik_drone2]
Mode = Normal
Address = 127.0.0.1
Port = 12000
```

and a simple MAVSDK file to read drone id looks like:
```
#!/usr/bin/env python3

import asyncio
from mavsdk import System

async def run():
    drone = System()
    await drone.connect(system_address="udp://:11000")

    drone2 = System()
    await drone2.connect(system_address="udp://:12000")

    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print(f"Drone discovered with UUID: {state.uuid}")
            break

    print("Waiting for drone2 to connect...")
    async for state in drone2.core.connection_state():
        if state.is_connected:
            print(f"Drone2 discovered with UUID: {state.uuid}")
            break

    # version_info = await drone.info.get_version()
    # print(version_info)

    id_info = await drone.info.get_identification()
    print(id_info)

    id_info2 = await drone2.info.get_identification()
    print(id_info2)

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    loop.run_until_complete(run())
```

But the output I get is the id of only a single drone.

Hence, how should I pair the UDP port with serial port so that when I connect a UDP in MAVSDK I get data of that particular drone only.

andreastoellefsen, 2020-11-27:
Can you get them to work one at a time?
if yes:
   Verify different sys_id (I assume you have already done this)
   Try to give endpoints non-similar names
   Try to start two separate mavlink-routerd and have them point to each other. you might have to use mode = eavesedropping.
ir no:
    something wrong other than mavlink-router

hashimKAifi, 2020-11-29:
> Can you get them to work one at a time?
> if yes:
> Verify different sys_id (I assume you have already done this)
> Try to give endpoints non-similar names
> Try to start two separate mavlink-routerd and [...]

Hi @andreastoellefsen yes i'm able to get them work one at a time.
I also tried giving different name to all endpoints. Code below
```
[General]

[UartEndpoint sik_drone1]
Device = /dev/ttyUSB0
Baud = 57600

[UdpEndpoint udp_of_sik_drone1]
Mode = Normal
Address = 127.0.0.1
Port = 11000

[UartEndpoint sik_drone2]
Device = /dev/ttyACM0
Baud = 57600

[UdpEndpoint udp_of_sik_drone2]
Mode = Normal
Address = 127.0.0.1
Port = 12000
```

The mavlink router using this config file runs without errors. But, it randomly sends telem data from Uart to Udp. Sometimes drone1 telem data to both udp endpoints, sometimes different, sometimes drone2 telem data to both udp endpoints. There is no control as in to specifically map drone1 telem data to a particular udp endpoint (say udp_of_sik_drone1).
Then, I tried to start 2 seperate mavlink-routerd but I get this error when I start the second mavlink-routerd (Could not bind to tcp socket (Address already in use) )

Thanks

jbeyerstedt, 2021-02-02:
First of all, I like to clarify how mavlink-router works. All telemetry messages received by one endpoint will get routed to all other endpoints (generally speaking). So both UDP endpoints will get the telemetry streams from both vehicles. (Commands addressed to a specific system ID will be routed to just one endpoint, though.)

So as far as I can tell, there are two options now:
1. Just use one UDP endpoint to connect to your application (and two UART endpoints to connect both drones). The only issue is, that MAVSDK-Python doesn't seem to support multiple vehicles (see https://github.com/mavlink/MAVSDK-Python/issues/102). (It's possible in C++ though to get a list of all discovered systems. I'm using it in one project.)
2. Use two mavlink-router instances (one for each UART-UDP pair) and your current code should work just fine. But then you wouldn't need mavlink-router because MAVSDK-Python can also connect to serial.

jbeyerstedt, 2021-07-28:
Closing because of inactivity.

Summary of the problem: Receive the data of two different drones (connected via two UART endpoints) with MAVSDK Python. Multiple drones on one link are not supported (yet) by MAVSDK-Python. So it was requested to have mavlink-router only route packets between certain endpoints (a UART and UDP pair each), like VLANs, which is also not currently implemented.

Workaround: Using multiple mavlink-router instances for the different communication links.

An outgoing filter by sysID on each endpoint was proposed in #257.

closed 2021-07-28 by jbeyerstedt