mavlink-router/mavlink-router #243: Heartbeat decode

galdalali, 2020-12-27:
Hi all,

I open'd up an issue on "yaapu/FrskyTelemetryScript" but maybe I can find a solution here..
#https://github.com/yaapu/FrskyTelemetryScript/issues/81

jbeyerstedt, 2021-02-10:
Do I understand it correctly, that you want a filter to just get heartbeats from a certain component ID? Then PR #238 might solve your issue (just see the last 2 commits).

galdalali, 2021-02-11:
That's great.
Can I suggest a different way to filter the messages? 
By messages you don't wanna reactive at this endpoint.
It's easy this way to determine which messages I don't want to receive in order to solve one endpoint decoding Heartbeat's from different clients (like YAAPU) and in order to reduce RF link bitrate to a target value.

jbeyerstedt, 2021-04-28:
I just took a look at your issues again to try to understand the root cause rather than just patching symptoms.

Your issue is, that the YAPPU software is receiving data from systems and/or components, you don't want it to receive. In you case, it does not only receive messages from the FMU, but also mission planner.
So the troublesome communication path is from mission planner via telemetry radio 2 to the companion running mavlink-router. This has to forward messages to the FMU, which has a second telemetry radio connected to a second UART port to talk to YAPPU.

So the issue is actually, that your FMU is forwarding data received on one telemetry port to the other one. Discarding messages from mission planner at mavlink-router would solve that issue, but then the communication between mission planner and the FMU wouldn't work any more. (The same applies to the data from your Comp 154 connected to the FMU via the companion, if the FMU needs this data.)

So at least with your current communication network, mavlink-router can't solve your problem.

galdalali, 2021-05-03:
Yea I know that patching the Mavlink router for this (The issue is the heartbeat from 154 ID to Yappu telemetry) it's least right way to solve that problem. 
There's couple of problem here.. first, I can not cancel the heartbeat from154 ID (although they have an option for setting heartbeat rate to 0 - Gremsy T3V3). Second, Yappu receive heartbeats from everyone without looking at that autopilot field and then decode this message and use it for motor and mode state which is scary as hell. 
I tried to leverage this issue just for adding filter by ID to Mavlink router since I think that's a must have capability

closed 2022-01-28 by lucasdemarchi