mavlink-router/mavlink-router #244: Stability issues

galdalali, 2021-01-11:
Hey there,

I'm having stability issues trying to run mavlink-router with 2 serial ports..
I'm running mavlink-router as a service in ubuntu with the following config:

```
[UartEndpoint FMU]
Device = /dev/ttyTHS1
Baud = 115200
FlowControl = 0

[General]
#Mavlink-router serves on this TCP port
TcpServerPort = 0
ReportStats = false
MavlinkDialect = auto

[UdpEndpoint VideoCore]
Mode = Normal
Address = 127.0.0.1
Port = 14553
ReportStats = true

[UdpEndpoint GCS]
Mode = Normal
# Update Me
Address = 192.168.1.203
# Update Me
Port = 14551
ReportStats = true

[UdpEndpoint MP]
Mode = Normal
# Update Me
Address = 192.168.1.203
# Update Me
Port = 14552
ReportStats = true

[UartEndpoint Gimbal]
Device = /dev/ttyTHS0
Baud = 115200
FlowControl = 0
```

Now sometimes it works perfectly and sometimes I get pull error right on startup and I can not put my finger on why.

```
Jan 10 05:24:44 localhost mavlink-routerd.sh[6151]: Open UDP [4] 127.0.0.1:14553
Jan 10 05:24:44 localhost mavlink-routerd.sh[6151]: Open UDP [5] 192.168.1.203:14551
Jan 10 05:24:44 localhost mavlink-routerd.sh[6151]: Open UDP [6] 192.168.1.203:14552
Jan 10 05:24:44 localhost mavlink-routerd.sh[6151]: Error while trying to write serial port latency: Unknown error -1
Jan 10 05:24:44 localhost mavlink-routerd.sh[6151]: Open UART [7] /dev/ttyTHS1 *
Jan 10 05:24:44 localhost mavlink-routerd.sh[6151]: UART [7] speed = 115200
Jan 10 05:24:44 localhost mavlink-routerd.sh[6151]: Error while trying to write serial port latency: Unknown error -1
Jan 10 05:24:44 localhost mavlink-routerd.sh[6151]: Open UART [8] /dev/ttyTHS0 *
Jan 10 05:24:44 localhost mavlink-routerd.sh[6151]: UART [8] speed = 115200
Jan 10 05:24:46 localhost mavlink-routerd.sh[6151]: poll error for fd 8, closing it
```

Sometimes I even start getting this nonstop warnings on startup and even after couple of minutes of normal operation and then I lose my connection to [UartEndpoint Gimbal]:

`log_warning("%u messages to unknown endpoints in the last %d seconds", _errors_aggregate.msg_to_unknown, LOG_AGGREGATE_INTERVAL_SEC);`

Anybody has an idea why?

lucasdemarchi, 2021-01-13:
what's the output of `stty -F /dev/ttyTHS0 raw low_latency`? And what board is that?

lucasdemarchi, 2021-01-13:
So... from your logs it seems that maybe you are having a device disconnect. See the "poll error for fd 8, closing it" above.  When that happens mavlink-router removes that system/component from its routing table. But the other systems are still sending messages to it.

galdalali, 2021-01-13:
> what's the output of `stty -F /dev/ttyTHS0 raw low_latency`? And what board is that?

The board is [Jetson Xavier-NX](https://www.nvidia.com/en-us/autonomous-machines/embedded-systems/jetson-xavier-nx/) on a [connect tech Quark](https://connecttech.com/product/quark-carrier-nvidia-jetson-xavier-nx-nano/) Carrier board.
I will tell you later on what is the output.

galdalali, 2021-01-13:
> So... from your logs it seems that maybe you are having a device disconnect. See the "poll error for fd 8, closing it" above. When that happens mavlink-router removes that system/component from its routing table. But the other systems are still sending messages to it.

Yes but why? It happens randomly and right on start. Most of the time it starts up OK.

lucasdemarchi, 2021-01-13:
> Yes but why? It happens randomly and right on start. Most of the time it starts up OK.

it may be a kernel driver issue for the serial port, it may be that the the device is disconnected, etc...  When it fails, can you still access the serial port from the terminal?

galdalali, 2021-01-16:
> what's the output of `stty -F /dev/ttyTHS0 raw low_latency`?

```
nvidia@localhost:~/setupscripts/videocore$ stty -F /dev/ttyTHS0 raw low_latency
stty: invalid argument ‘low_latency’
Try 'stty --help' for more information.
```
That's the output of `sudo stty -F /dev/ttyTHS0 -a`

```
speed 115200 baud; rows 0; columns 0; line = 0;
intr = ^C; quit = ^\; erase = ^?; kill = ^U; eof = ^D; eol = <undef>; eol2 = <undef>; swtch = <undef>; start = ^Q; stop = ^S; susp = ^Z; rprnt = ^R; werase = ^W; lnext = ^V; discard = ^O; min = 0; time = 0;
-parenb -parodd -cmspar cs8 -hupcl -cstopb cread clocal -crtscts
-ignbrk -brkint -ignpar -parmrk -inpck -istrip -inlcr -igncr -icrnl -ixon -ixoff -iuclc -ixany imaxbel -iutf8
-opost -olcuc -ocrnl -onlcr -onocr -onlret -ofill -ofdel nl0 cr0 tab0 bs0 vt0 ff0
-isig -icanon -iexten -echo -echoe -echok -echonl -noflsh -xcase -tostop -echoprt -echoctl -echoke -flusho -extproc
```

By the way, That's a proper run, everything works:

```
 mavlink-router.service - MAVLink Router
   Loaded: loaded (/lib/systemd/system/mavlink-router.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2021-01-16 16:22:08 UTC; 6min ago
  Process: 5963 ExecStartPre=/bin/sleep 4 (code=exited, status=0/SUCCESS)
 Main PID: 6597 (mavlink-routerd)
    Tasks: 2 (limit: 4915)
   CGroup: /system.slice/mavlink-router.service
           ├─6597 /bin/bash /usr/local/bin/mavlink-routerd.sh
           └─6715 /usr/local/bin/mavlink-routerd -c /usr/local/bin/mavlink-router_oper.conf

Jan 16 16:22:08 localhost mavlink-routerd.sh[6597]: NVPM WARN: patched tpc_pg_mask: 0x4
Jan 16 16:22:10 localhost mavlink-routerd.sh[6597]: Open UDP [4] 127.0.0.1:14553
Jan 16 16:22:10 localhost mavlink-routerd.sh[6597]: Open UDP [5] 192.168.1.114:14551
Jan 16 16:22:10 localhost mavlink-routerd.sh[6597]: Open UDP [6] 192.168.1.114:14552
Jan 16 16:22:10 localhost mavlink-routerd.sh[6597]: Error while trying to write serial port latency: Unknown error -1
Jan 16 16:22:10 localhost mavlink-routerd.sh[6597]: Open UART [7] /dev/ttyTHS1 *
Jan 16 16:22:10 localhost mavlink-routerd.sh[6597]: UART [7] speed = 115200
Jan 16 16:22:10 localhost mavlink-routerd.sh[6597]: Error while trying to write serial port latency: Unknown error -1
Jan 16 16:22:10 localhost mavlink-routerd.sh[6597]: Open UART [8] /dev/ttyTHS0 *
Jan 16 16:22:10 localhost mavlink-routerd.sh[6597]: UART [8] speed = 115200
```

galdalali, 2021-01-16:
> > Yes but why? It happens randomly and right on start. Most of the time it starts up OK.
> 
> it may be a kernel driver issue for the serial port, it may be that the the device is disconnected, etc... When it fails, can you still access the serial port from the terminal?

Yep, restarting the mavlink-router and everything works fine

Vicidel, 2022-07-27:
Hello all

I have the same exact same issue, using `/dev/ttyACM0`with mavlink-router. Did anyone figure out a solution? Below the terminal output and my conf file.

Victor.

```
$ mavlink-routerd 
mavlink-router version v2-229-g022333d
Error while trying to write serial port latency on /dev/ttyACM0: Operation not supported
Opened UART [4]droneEndpoint: /dev/ttyACM0
UART [4]droneEndpoint: speed = 3000000
Opened UDP Client [5]mavrosEndpoint: 127.0.0.1:14540
poll error for fd 4
Critical fd 4 got error, exiting
```

```
$ cat /etc/mavlink-router/main.conf 
[General]
Log = /home/rigi/shared_data/logs
LogMode = while-armed
TcpServerPort = 0
MavlinkDialect = auto
DebugLogLevel = info

[UartEndpoint droneEndpoint]
Device = /dev/ttyACM0
Baud = 3000000

[UdpEndpoint mavrosEndpoint]
Mode = Normal
Address = 127.0.0.1
Port = 14540
```