mavlink-router/mavlink-router #245: Build on Focal fails with "possibly undefined macro: AC_CHECK_LIB"

peterbarker, 2021-01-12:
```
apsync@apsync:~/GitHub/mavlink-router$ ./autogen.sh
libtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, 'build-aux'.
libtoolize: linking file 'build-aux/ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: linking file 'm4/libtool.m4'
libtoolize: linking file 'm4/ltoptions.m4'
libtoolize: linking file 'm4/ltsugar.m4'
libtoolize: linking file 'm4/ltversion.m4'
libtoolize: linking file 'm4/lt~obsolete.m4'
configure.ac:57: error: possibly undefined macro: AC_CHECK_LIB
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
autoreconf: /usr/bin/autoconf failed with exit status: 1
apsync@apsync:~/GitHub/mavlink-router$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.1 LTS
Release:	20.04
Codename:	focal
apsync@apsync:~/GitHub/mavlink-router$ 
```

Script being used to build (from the apsync repo):
```
pushd ~/GitHub
 rm -rf mavlink-router
 git clone --recurse-submodules https://github.com/01org/mavlink-router
 pushd mavlink-router
  git submodule update --init --recursive
  ./autogen.sh && ./configure CFLAGS='-g -O2' \
            --sysconfdir=/etc --localstatedir=/var --libdir=/usr/lib64 \
            --prefix=/usr
  time make
  sudo make install
 popd
popd

```

lucasdemarchi, 2021-01-12:
You need to install pkg-config.

Unrelated, but you also need to update mavlink-router repo location in the script above.

peterbarker, 2021-01-12:
Thanks @lucasdemarchi - will update the repo and the script.

closed 2021-01-12 by peterbarker