mavlink-router/mavlink-router #246: build error

helloEarth7, 2021-01-15:
HI,

I tried to build mavlink-router from yesterday on unbuntu 18.04
I've got 2 error while build.

make --no-print-directory all-am
  CXX      src/mavlink-router/binlog.o
src/mavlink-router/binlog.cpp: In member function ‘virtual int BinLog::write_msg(const buffer*)’:
src/mavlink-router/binlog.cpp:104:34: error: ‘const mavlink_msg_entry_t {aka const struct __mavlink_msg_entry}’ has no member named ‘msg_len’
     if (payload_len > msg_entry->msg_len) {
                                  ^~~~~~~
src/mavlink-router/binlog.cpp:105:34: error: ‘const mavlink_msg_entry_t {aka const struct __mavlink_msg_entry}’ has no member named ‘msg_len’
         payload_len = msg_entry->msg_len;
                                  ^~~~~~~
Makefile:1006: recipe for target 'src/mavlink-router/binlog.o' failed
make[1]: *** [src/mavlink-router/binlog.o] Error 1
Makefile:760: recipe for target 'all' failed
make: *** [all] Error 2

lucasdemarchi, 2021-01-15:
do you have all the submodules updated?

helloEarth7, 2021-01-15:
@lucasdemarchi Yes I did 
$ git submodule update --init --recursive

$ sudo apt install python-future
$ sudo apt install python3-future
$ sudo apt install libtool
$ sudo apt install autoconf

helloE9, 2021-01-15:
@lucasdemarchi
The error caused by I cloned src to sub-directory of firmware.
BUild was successful.

Thanks.
case closed

closed 2021-01-15 by helloE9

lucasdemarchi, 2021-01-15:
What do you mean by "firmware"?  It would be strange to have a problem depending on the directory you cloned the repo.

helloE9, 2021-01-15:
@lucasdemarchi
I cloned the directory of marvlink-router in the directory where I cloned the px4's firmware, but there was a build error.
When I cloned back to the upper directory and built, the problem was solved.
Tomorrow, I plan to install it on a raspberry.
My purpose is to use 2 QGCs linked by wifi.