mavlink-router/mavlink-router #248: partial log

Dan-Moss, 2021-01-20:
streaming logs from cube orange to companion with:
```
[General]  
ReportStats=false 
MavlinkDialect=ardupilotmega 
RetryTimeout=10
DebugLogLevel=info
Log=/log/flight
LogMode = while-armed
```
I get a log for each flight, however some are seemingly randomly cut short. I have checked kernal logs for hardware disconnect and that is not the issue.
I see this error:
```
lseek failed (Bad file descriptor)
Restarting log...
BinLog not started
```
but does not seem to match up with the shortened logs. I've read somewhere this is to do with trying to write to a closed file?
[mavlink_router.zip](https://github.com/mavlink-router/mavlink-router/files/5841387/mavlink_router.zip)

[image]

i have also checked there is no failsafes or other events from the SD log.

any help in going about debugging this?

Dan-Moss, 2021-01-21:
unsure if this build error has anything to do with it?
```
  CXX      src/mavlink-router/binlog.o
In file included from /mavlink-router/src/mavlink-router/binlog.cpp:26:0:
/mavlink-router/src/mavlink-router/binlog.cpp: In member function 'void BinLog::_logging_data_process(mavlink_remote_log_data_block_t*)':
/mavlink-router/src/common/log.h:57:63: warning: format '%ld' expects argument of type 'long int', but argument 3 has type 'ssize_t {aka int}' [-Wformat=]
 #define log_error(...) Log::log(Log::Level::ERROR, __VA_ARGS__)
                                                               ^
/mavlink-router/src/mavlink-router/binlog.cpp:209:9: note: in expansion of macro 'log_error'
         log_error("Log partial write %ld", r);
         ^
  CXX      src/common/conf_file.o
  CXX      src/mavlink-router/endpoint.o
/mavlink-router/src/mavlink-router/endpoint.cpp: In member function 'virtual void Endpoint::print_statistics()':
/mavlink-router/src/mavlink-router/endpoint.cpp:392:88: warning: format '%lu' expects argument of type 'long unsigned int', but argument 4 has type 'uint64_t {aka long long unsigned int}' [-Wformat=]
            (_stat.read.crc_error * 100) / read_total, _stat.read.crc_error_bytes / 1000);
                                                       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
/mavlink-router/src/mavlink-router/endpoint.cpp:395:94: warning: format '%lu' expects argument of type 'long unsigned int', but argument 3 has type 'uint64_t {aka long long unsigned int}' [-Wformat=]
     printf("\n\t\tHandled: %u %luKBytes", _stat.read.handled, _stat.read.handled_bytes / 1000);
                                                               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
/mavlink-router/src/mavlink-router/endpoint.cpp:399:84: warning: format '%lu' expects argument of type 'long unsigned int', but argument 3 has type 'uint64_t {aka long long unsigned int}' [-Wformat=]
     printf("\n\t\tTotal: %u %luKBytes", _stat.write.total, _stat.write.bytes / 1000);
                                                            ~~~~~~~~~~~~~~~~~~~~~~~~^
  CXX      src/common/log.o
  CXX      src/mavlink-router/logendpoint.o
In file included from /mavlink-router/src/mavlink-router/logendpoint.cpp:38:0:
/mavlink-router/src/mavlink-router/logendpoint.cpp: In member function 'void LogEndpoint::_delete_old_logs()':
/mavlink-router/src/common/log.h:53:63: warning: format '%lu' expects argument of type 'long unsigned int', but argument 3 has type 'uint64_t {aka long long unsigned int}' [-Wformat=]
 #define log_debug(...) Log::log(Log::Level::DEBUG, __VA_ARGS__)
                                                               ^
/mavlink-router/src/mavlink-router/logendpoint.cpp:127:5: note: in expansion of macro 'log_debug'
     log_debug("[Log Deletion]  Total free space: %lumb. Min free space: %lumb",
     ^
  CXX      src/mavlink-router/main.o
```

Dan-Moss, 2021-02-04:
https://github.com/mavlink-router/mavlink-router/issues/239
https://github.com/mavlink-router/mavlink-router/issues/155
similar issues i beleive

Dan-Moss, 2021-02-08:
@lucasdemarchi @peterbarker any help/ideas on how I can go about debugging this?

peterbarker, 2021-02-08:
@Dan-Moss bisect the problem by using MAVProxy's `dataflash_logging` module - that will tell you if the problem really is with mavlink-router.

The error hints that it is.  Check the `dmesg` output to make sure you're not getting errors on your filesystem.

Failing that - time to look at the mavlink-router code, tracing all places where that file descriptor is used (and perhaps inserting some debug at each point to see if something is being crossed when it shouldn't be).

Dan-Moss, 2021-02-16:
hi @peterbarker , here is a debug output snapshot of a log file that was saved as 0 bytes, but as you can see data is there.
normally successful files are read only, but the erroneous one was read and write still at 0 byte. maybe linked or 2 issues at play. see my post a couple of comments up for other people with a similar issue. 

[mavlink_router_0_byte](https://github.com/mavlink-router/mavlink-router/files/5987773/mavlink_router_pi2_February.9.2021.4-17.PM.zip)

a partial log, from the bottom up. not sure where it started/stopped.
[partial_log.zip](https://github.com/mavlink-router/mavlink-router/files/5987962/partial_log.zip)

in regards to the `dmesg` :
`[   55.966745] uart-pl011 3f201000.serial: no DMA platform data` i see this and dont understand if it enough to know if relevent, but the FCU is still connected and getting data.
and `uart-pl011 3f201000.serial: cts_event_workaround enabled` on boot, but nothing showing a disconnect or error.

peterbarker, 2021-03-30:
On Tue, 30 Mar 2021, Dan wrote:

> hi @peterbarker it seems that this is only occurring on flight controllers
> IN standby mode, can you think why this might be the cause of the issue?

Ah.  Pretty sure mavlink-router can handle two different protocols; 
ArduPilot's logging-over-mavlink and PX4 Firmware's logging-over-mavlink.

ArduPilot doesn't have a "standby" mode - so I think you must be refering 
to PX4 Firmware.

While I've implemented the client-side of ArduPilot's protocol several 
times, I've not implemented PX4 Firwmare's - and I'm the author of neither 
in mavlink-router :-)  I'm also not a PX4 developer and have very limited 
knowledge of that codebase.

Sorry...

Dan-Moss, 2021-03-31:
we are using ArduCopter. Standby mode was in from https://github.com/ArduPilot/ardupilot/pull/12441

unsure what impact this may have...

peterbarker, 2021-03-31:
On Wed, 31 Mar 2021, Dan wrote:

> we are using ArduCopter. Standby mode was in from ArduPilot/ardupilot#12441

Ah.  We never should have attached the word "mode" to that functionality 
:-)

> unsure what impact this may have...

It probably shouldn't have any.

lucasdemarchi, 2021-04-27:
```
lseek failed (Bad file descriptor)
Restarting log...
BinLog not started
```

Humn... so here what seems to be happening is that some other component is enabling logging. Then we are receiving a packet before trying to start . This can be seen by the file descriptor being invalid (EBADFD) and then stop() failing with "Binlog not started".

On a hindsight those message should be made much clearer. And we should handle the case logging had already started when mavlink-router got up.

Questions: do you have other components possibly starting/stopping the logging? Do you have mavlink-routerd being restarted automatically when if exits/fails?  This second case I'm thinking if mavlink-router was somehow failing, then restarted and upon restart the flight stack logging was already going on. The fact that we don't serialize the "last seqno acked" to recover, probably means we would lose a portion of the log until we resync.  I fixed a bug today #263  regarding  mavlink-routerd exiting when we got a TCP disconnection - do you have TCP connections going on?

Dan-Moss, 2021-04-29:
>  do you have other components possibly starting/stopping the logging? Do you have mavlink-routerd being restarted automatically when if exits/fails?

No, only started once at boot time and not restarted

> do you have TCP connections going on?

fcu <> companion is serial, then `value="tcp://0.0.0.0:5760"/` for mavlink router

lucasdemarchi, 2021-04-29:
Ok. Can you check if master branch improves that?  Previously if we had a problem in a TCP endpoint we would stop mavlink-router

Dan-Moss, 2021-04-30:
yes, will check and report back 👍 thanks

Dan-Moss, 2021-05-05:
hi @lucasdemarchi since the change i now sometimes get:
```
Open UART [4] /dev/ttyAMA0 *
UART [4] speed = 921600
Open TCP [5] 0.0.0.0:5760 *
TCP connection [7] accepted
poll error for fd 7
BinLog not started
TCP Connection [7] closed
```

Dan-Moss, 2021-05-26:
@lucasdemarchi any insights into my last post?

Dan-Moss, 2021-06-01:
went through a recent set of logs, pairing up the ones that were cut short. no errors on STD out, gets piped to the file in the screenshot shown
the one in question just stops logging to the companion computer after 5s
fine compared to the other FCU
[image]

peterbarker, 2021-06-01:
On Tue, 1 Jun 2021, Dan wrote:

> went through a recent set of logs, pairing up the ones that were cut short. no errors on STD out, gets piped to the file in the screenshot shown
> the one in question just stops logging to the companion computer after 5s
> fine compared to the other FCU
> image

Attaching strace to the process and waiting for a failure might be your 
best way forward here.