mavlink-router/mavlink-router #249: UDP packet looping with same payload between 2 PCs with mavlink-routerd

mjmlopes, 2021-01-21:
Hello all.

From time to time, QGroundControl (v.4.1.0) looses communication with Pixhawk 4 (HW PX4_FMU_V5, FW v.1.11.3). In order to have communication again, one of the mavlink-routerd (git 51c72ff) must be terminated and then launched.

Scenario:
Pixhawk 4 TELEM1 port is connected to an Serial-USB converter without HW flow control. (same happens if PC is connected directly to Pixhawk 4 USB port)
Serial-USB converter connects to a PC running mavlink-routerd in Ubuntu (20.4.1), master UART 57.6k and UDP endpoint "normal" 10.0.0.2:24550. This PC (10.0.0.1) is connected to another PC (10.0.0.2) via ethernet.
The 2nd PC runs mavlink-routerd in Debian 10, master UDP 0.0.0.0:24550 and UART endpoint 57.6k.
The 3rd PC runs QGroundControl in Ubuntu (20.4.1) with UART set to 57.6k.

When both mavlink-routerd instances are active, the following message is repeatedly thrown on both terminals:
> MAV_CMD_LOGGING_START result(2) is different than accepted

Pixhawk nsh terminal repeatedly shows:
> WARN  [mavlink] Not enough bandwidth to enable log streaming

After some (random) time, QGroundControl sensor data seems sluggish for one or two and then stops with "COMMUNICATION LOST" message being printed on the screen.
At around the same time, on Pixhawk nsh terminal the message appears:
> WARN  [commander] Connection to ground station lost

At around the same time, both mavlink-routerd instances throws to terminal the following (number of "incomplete messages" varies):
> Endpoint UART [4]: 902 incomplete messages in the last 5 seconds

Performing a tcpdump shows that mavlink UDP packets are being exchanged with the same payload multiple times. Different payloads goes back and forth multiple times. UDP payload (MAVLink packet) example:

> fd1c0000b201b3010124000058af7b918403840384038403e485fd110000b401010100002f002612

Interesting though, if QGroundControl serial port speed is increased, the communication to pixhawk became intermittent and less "incomplete messages" are present on logs.

This issue doesn't rise if 3rd PC runs QGroundControl in UDP server mode and 1st PC mavlink is changed to have the UDP endpoint as the 3rd PC.

No relevant messages were seen on syslog on all PCs.

Thanks.

jbeyerstedt, 2021-03-17:
I don't quite understand, how the third PC (QGC) is connected to the second PC. Your description looks like UART, but why? The easiest way would be to just let the second mavlink-router send data via UDP to port 14550 on the third PC (if they can reach each other via UDP/IP obviously).

Mavlink-router is very quiet when you don't add the `-v` verbose flag, so to debug this issue further, you might have to take a look at that output. One thing to take a look at would be the list or "known components" in each endpoint. It should just have one entry on each endpoint (either the FMU or QGC (255/160 by default)). If both systems are at one endpoint, there's a message loop somewhere and the communication will come to a stop.

Otherwise, I would need the config files or CLI parameters you used to further investigate. I always get a bit confused with specifying the endpoints via the CLI, because I just use a config file everywhere (easier to deploy/ maintain when started with systemd).

jbeyerstedt, 2021-09-09:
Closing because of inactivity. Please comment, if this should be re-opened.

closed 2021-09-09 by jbeyerstedt