mavlink-router/mavlink-router #250: Proposal: New config file format

jbeyerstedt, 2021-02-09:
Judging from some issues and my own experience, especially the "mode" parameter of the UDP endpoint leads to confusion. With regards to some of my outstanding pull requests, an overhaul of the configuration file might lead to a more intuitive understanding and better extendability.

I propose to to use https://toml.io/ as configuration file language, because it's well specified and has parsers for C++ (and many other languages).

## Changes to the Configuration Options
The endpoint names aren't needed by the syntax, but would make the debug output easier to understand, when the currently used number was replaced with the custom name.

### UDP Endpoint
An UDP endpoint will drop the mode field and instead use a listening or a target address. Depending on which of these is defined, the endpoint will fall into sending (normal) or listening (eavesdropping) mode.
Additionally, I would combine the IP address and port, so that the syntax is the same, as in the command line options. 
```toml
# "Normal" UDP endpoint
[[endpoints.udp]]
target-address = "127.0.0.1:14550"
```
```toml
# "Eavesdropping" UDP endpoint
[[endpoints.udp]]
listening-address = "0.0.0.0:14550"
```

### TCP Server
The old `TcpServerPort` will be moved to the endpoints section of the config file, because it's basically an endpoint as well. (Actually a dynamically created one for each connected client.)
```toml
[endpoints.tcp-server]
port = 5760
# TODO: is a listening address needed?
```
The general section might be a better place for a command and control server's port (or unix domain socket), as proposed in #242.

### Message Filtering
I renamed the filter parameter to "outgoing message ID filter", because it's a more precise description. The parameter key `filter.out.msgID` will allow for filtering on the incoming path (before the message is presented to all other endpoints), when it's needed, as well as having other types of filters, e.g. by component ID (see #238).

## Configuration file example
```toml
[general]
# mavlink-dialect = "auto"  # options: auto (default), common or ardupilotmega
# debug-loglevel = "info"   # options: error, warning, info (default) or debug

[general.messagelog]
# directory = "/foo/bar"  # store flight stack logs here, otherwise disabled
# mode = "always"         # options: always (default), while-armed
# min-free-space = 0      # if >0: delete old log files, when less than x bytes are available on the disk
# max-files = 0           # if >0: delete old log files to keep the number below this

[[endpoints]]
# e.g. de-duplication options

[endpoints.tcp-server]
port = 5760
# TODO: is a listening address needed?

[[endpoints.uart]]
name = "fmu"
device = "/dev/ttyS0"
# baud = 115200
# flow-control = false         # options: Enable UART flow control (default: disabled)
# filter.out.msgID = [ 0, 1 ]  # optional, just send out this list of message IDs

[[endpoints.udp]]
name = "udp1"
target-address = "127.0.0.1:14550"  # send messages to this address and port, receive on random port after first outgoing message
# filter.out.msgID = [ 0, 1 ]       # optional, just send out this list of message IDs

[[endpoints.udp]]
name = "udp2"
listening-address = "127.0.0.1:14551"  # listen on this interface and port (0.0.0.0 for all interfaces), send to last connected client (after first incoming message)
filter.out.msgID = [ 0, 1 ]            # optional, just send out this list of message IDs

[[endpoints.udp]]
name = "udp-ipv6"
target-address = "[::1]:14552"  # IPv6 addresses must be enclosed in square brackets, UDP multicast is also possible
# filter.out.msgID = [ 0, 1 ]   # optional, just send out this list of message IDs

[[endpoints.tcp]]
name = "tcp1"
target-address = "127.0.0.1:25790"
RetryTimeout = 5               # seconds to wait to try to re-connect in case of lost connection, 0 to disable reconnection
# filter.out.msgID = [ 0, 1 ]  # optional, just send out this list of message IDs
```

lucasdemarchi, 2021-02-10:
It's close enough to an ini file like we have - I don't feel like adding the dependency on their library for a very small change. The changes you are proposing are more on the semantic, that could be done on the current format as well, but it may cause more harm than benefit.

Btw, the format in use is heavily inspired by systemd, which is the current init for most Linux distributions.

jbeyerstedt, 2021-12-22:
Closing, because we don't want to break backwards compatibility.

closed 2021-12-22 by jbeyerstedt