mavlink-router/mavlink-router #251: Message filtering increases "unknown endpoint" error counter

jbeyerstedt, 2021-02-10:
Setting a filter in an endpoint, will lead to an increased "unknown message" error counter. I think, this shouldn't be the case, because filtering out the message is an intended rejection by the endpoint (in contrast to the a message, which is rejected by all endpoint because of it's target system ID).

With this example config:
```
[General]
TcpServerPort=5790
ReportStats=false
MavlinkDialect=common

[UdpEndpoint d2x]
Mode = Normal
Address = 127.0.0.1
Port = 14500
Filter = 0, 1

[UartEndpoint uart4]
Device = /dev/ttymxc3
Baud = 115200
```

The verbose console log will show something like this (my drone has SysID 23 and QGC is running behind the UDP endpoint):
```
UDP [4] got 21 bytes
Endpoint [5] got message 0 to -1/-1 from 255/190
	Known components:
		23/1
Endpoint [5] accepted message 0 to -1/-1 from 255/190
UART [5] wrote 21 bytes
UART [5] got 30 bytes
Endpoint [4] got message 74 to -1/-1 from 23/1
	Known components:
		255/190
Message 74 to unknown sysid/compid: 4294967295/4294967295
UART [5] got 40 bytes
Endpoint [4] got message 30 to -1/-1 from 23/1
	Known components:
		255/190
Message 30 to unknown sysid/compid: 4294967295/4294967295
```

And in between (or without the `-v` flag), there will be these messages:
```
62 messages to unknown endpoints in the last 5 seconds
```

jbeyerstedt, 2021-02-10:
Would it be fine to change the `return false` to `return true` in [line 336 of src/mavlink-router/endpoint.cpp](https://github.com/mavlink-router/mavlink-router/blob/master/src/mavlink-router/endpoint.cpp#L336) or is there is side-effect, which I don't have in mind?

jkflying, 2021-02-10:
How about instead getting `accept_msg` to return an enum instead of a bool? Then depending on the enum you know whether it was an unknown message, parsed but rejected message, etc.

jbeyerstedt, 2021-02-10:
I haven't looked close enough again. `write_msg()` is actually a method of the Mainloop at that place and handles some more things about blocking endpoints. So the enum will be the less invasive approach.

commit 56a37f4c (2021-12-04): endpoint: Do not increase msg to unknown cnt on filter (fix #251)
closed 2021-12-04 by lucasdemarchi