mavlink-router/mavlink-router #252: Closed connection to TCP port crashes mavlink-router

jbeyerstedt, 2021-02-12:
I was trying to use pymavlink to analyze a MAVLink message. To get the message, I tried to connect to the TCP port (`TcpServerPort` parameter) of mavlink-router.

But when my python script terminates, mavlink-router will stop working, because it crashed after the TCP connection was lost. As far as I understood, the TCP server should be able to even handle multiple clients, so one client disconnecting shouldn't lead to mavlink-router terminating, right?

The verbose output looks like this (output shortened):
```
[...]

TCP connection [9] accepted
Accepted TCP connection on [9]
UDP [5] got 30 bytes
Endpoint [4] got message 74 to -1/-1 from 23/1
	Known components:
		23/1
Endpoint [6] got message 74 to -1/-1 from 23/1
	Known components:
		255/190
Endpoint [6] accepted message 74 to -1/-1 from 23/1
UDP [6] wrote 30 bytes
Endpoint [9] got message 74 to -1/-1 from 23/1
	Known components:
Endpoint [9] accepted message 74 to -1/-1 from 23/1
TCP [9] wrote 30 bytes

[...]

UDP [4] got 13 bytes
Endpoint [5] got message 42 to -1/-1 from 23/1
	Known components:
		23/1
Endpoint [6] got message 42 to -1/-1 from 23/1
	Known components:
		255/190
Endpoint [6] accepted message 42 to -1/-1 from 23/1
UDP [6] wrote 13 bytes
Endpoint [9] got message 42 to -1/-1 from 23/1
	Known components:
Endpoint [9] accepted message 42 to -1/-1 from 23/1
TCP [9] wrote 13 bytes
UDP [5] got 13 bytes
poll error for fd 9, closing it
TCP Connection [9] closed
```

The second part of this issue is, that the return code is zero, so the systemd unit does not re-start mavlink-router after the crash.

lucasdemarchi, 2021-02-13:
> But when my python script terminates, mavlink-router will stop working, because it crashed after the TCP connection was lost. As far as I understood, the TCP server should be able to even handle multiple clients, so one client disconnecting shouldn't lead to mavlink-router terminating, right?

right, the bug sounds legit

> The second part of this issue is, that the return code is zero, so the systemd unit does not re-start mavlink-router after the crash.

yep, another bug

windelbouwman, 2021-03-05:
@jbeyerstedt I can confirm this issue exists. Do you have some minimal example script to trigger this bug?

koster1996, 2021-03-11:
I can also confirm this issue exists. I'm able to reproduce it using px4_sitl gazebo and QGroundControl.
It requires a PX4 development environment ([link](https://docs.px4.io/master/en/dev_setup/dev_env_linux_ubuntu.html)).

mavlink-router is configured as follows in `/etc/mavlink-router/main.conf`:
    `[UdpEndpoint SIM]`
    `Mode = Eavesdropping`
    `Address = 127.0.0.1`
    `Port = 14550`

To reproduce:
call `mavlink-routerd` in a terminal
call `make px4_sitl gazebo_iris` in the PX4-Autopilot folder

Then open QGroundControl, connect to the mavlink router over TCP on adress 127.0.0.01 on port 5760 using the "Comm Links" tab under "Application Settings".  Once QGroundControl connects, disconnecting it causes mavlink router to crash. 

The only output is `poll error for fd 7, closing it`

Curiously, using `mavlink-routerd -v` does not appear to have this issue

windelbouwman, 2021-03-11:
If the fact is that epoll returns an error, could you print out the error, to see what the error code is? Maybe it is a legitimate error which could occur and should be dealt with.

beenjamminnallday, 2021-03-17:
I am getting the same error. I am using my laptop to TCP on the mavlink-router (RasPi) which is connected by serial to my pixhawk. 

On my laptop, i'm running:
` "mavsdk_server_win32.exe" tcp://mavlink-routerIPaddress:5760`

On my Pi, I'm running:
`mavlink-routerd /dev/ttyAMA0:57600`

The connection works well and i'm able to send data and run py scripts from my laptop to my pixhawk.  However similarly to the issue described above, when I close the mavlink server on my laptop (closing the TCP connection), the mavlink-router crashes. 

My verbose output:
```
UART [4] got 8 bytes
Endpoint [4] got message 30 to -1/-1 from 1/1
        Known components:
                1/1
Endpoint [7] got message 30 to -1/-1 from 1/1
        Known components:
                245/190
Endpoint [7] accepted message 30 to -1/-1 from 1/1
TCP [7] wrote 40 bytes
UART [4] got 8 bytes
UART [4] got 8 bytes
UART [4] got 8 bytes
UART [4] got 8 bytes
UART [4] got 8 bytes
UART [4] got 8 bytes
poll error for fd 7, closing it
TCP Connection [7] closed
```

lucasdemarchi, 2021-03-17:
> poll error for fd 7, closing it

Looks like we are not handling disconnection well on TCP. This needs to be fixed. But...

> I am currently using my laptop to TCP on the mavlink-router (RasPi) which is connected by serial to my pixhawk & results in a similar error.

why are you running mavlink-router both on drone and your laptop. If you are using TCP, why not simply connect directly to the mavlink router on the drone?

beenjamminnallday, 2021-03-17:
I'm only running the router on the drone, on my laptop i'm running mavsdk-server.

windelbouwman, 2021-03-18:
It still is somewhat unclear to me how this `EPOLLERR` should be handled. The code path is shown here, and as can be seen, we request to exit mavlink-router when this occurs:

https://github.com/mavlink-router/mavlink-router/blob/master/src/mavlink-router/mainloop.cpp#L299

This post seems to give a hint on what to do:

https://stackoverflow.com/questions/24119072/how-to-deal-with-epollerr-and-epollhup

We could simply try to read / write when this error occurs, and the read / write code will detect the error, and invoke the appropriate actions.

Any other ideas what we should do when this error arises?

Another question I have is when this error arises, what triggers it?

highfreq, 2021-03-30:
Same issue that got closed i raised about 1 year ago:

https://github.com/mavlink-router/mavlink-router/issues/214

windelbouwman, 2021-03-30:
It is a somewhat spurious issue, that's probably why it took that long to re-appear. Did you find a fix for it?

highfreq, 2021-04-17:
Any chance this important bug will be corrected in future releases?

thanks

LorenzMeier, 2021-04-17:
I will close this issue and file a new one because it is not actually crashing - it is a somewhat debatable design choice to terminate the application (by design) when the port closes. That's unexpected and I agree that it would be better to avoid it.

highfreq, 2021-04-17:
Really do not understand the logic of closing it down when a tcp connection is closed. What harm can it do to gracefully handle the disconnection and leave the service up for a new connection. If it is not a bug is a really weird behaviour.

jbeyerstedt, 2021-04-17:
@LorenzMeier I can also change the title, if we want to debate about the wording. You're right, that it's not crashing actually (but that's one of the underlying issues).
But I thought, that the TCP port is meant to handle multiple clients, so disconnecting should be an expected and allowed behavior. So exiting returning 0, will stop any systemd-unit and leave the service unavailable (e.g. the one installed by default).

highfreq, 2021-04-17:
Actually i would prefer it crashing so it would restart :)

windelbouwman, 2021-04-17:
The shutting down of the entire router seems somewhat overkill to me. It would make sense to me if a connection broke down, that connection was dropped from a list, and the router would simply continue. Maybe a configuration could be added to trigger a crashing exit when a connection brakes down.

The problem I'm facing is that I have a broken serial port driver, which I open, but cannot open for the second time. In other words, the mavlink-router crashes, so I need to reboot the whole machine. This is undesirable.

highfreq, 2021-04-18:
We connect to it from remote on an LTE line to monitor telemetry, each time the conection is lost for a fraction of a second it crashes and there is no way to reconnect untill next take off.

lucasdemarchi, 2021-04-18:
I agree this is far from ideal. And it was not really the behavior intended. I will take a look soon to fix this bug.

lucasdemarchi, 2021-04-18:
This is actually a behavior that changed in 7e7ea939. I think we should leave at least TCP connections out of that check and maybe we should return an error code in that case, because then we may have a systemd configuration to take action on specific situations.

highfreq, 2021-04-18:
can't it just accept that a tcp connection can  be terminated an survive it keeping on listening if there are new connections coming in? Like every server does.

lucasdemarchi, 2021-04-18:
@highfreq  yes, that was the intention. There is a function to cleanup the dropped connections. When it was added support for dropping a serial in the commit mentioned above, the behavior was changed and impacted TCP connections as well. That needs to be fixed.

highfreq, 2021-04-18:
Thank you for stepping in and thank you for you interest in fixing the prob.

lucasdemarchi, 2021-04-26:
@highfreq  can you try #263 ?

highfreq, 2021-04-26:
Hello, thanks. Should i compile from master?

highfreq, 2021-04-26:
Just a quick question, how can i cleanly remove a previous version before compiling and install new one?

highfreq, 2021-04-26:
I guess i should compile tip-fix-exit.
Will do tomorrow, just would like to know how to cleanly deinstall previous before installing this.

lucasdemarchi, 2021-04-26:
not sure how you installed the first one... For testing you can it directly from the repo, no need to install it

highfreq, 2021-04-27:
This is how i compiled and installed on raspberry:
git clone https://github.com/intel/mavlink-router.git
cd mavlink-router
git submodule update --init --recursive
./autogen.sh
./configure CFLAGS='-g -O2' --sysconfdir=/etc --localstatedir=/var --libdir=/usr/lib --prefix=/usr
make
sudo make install

Since it is a test machine we use daily i would like to deinstall before testing new one.

lucasdemarchi, 2021-04-27:
You can just remove /usr/bin/mavlink-routerd and /usr/lib/systemd/system/mavlink-router.service, these are the only files we install. If you just install again it will override them.

lucasdemarchi, 2021-04-27:
btw you can check what we install by doing:  `make install DESTDIR=/tmp/test`. Everything we install will now be inside this /tmp/test directory

highfreq, 2021-04-27:
Great thanks, will do it today and let you know.

highfreq, 2021-04-27:
Just one more little thing, what would be the git line to download the correct branch?

git clone.........

highfreq, 2021-04-27:
Used:
git clone -b tip-fix-exit https://github.com/intel/mavlink-router.git
hope it is correct.
Than i compiled it and in stalled like the previous.

lucasdemarchi, 2021-04-27:
if you have it cloned already (and assuming "origin" is the name of your remote):

```
git fetch origin refs/pull/263/head:refs/heads/pr-263
git checkout pr-263
```

Or you can install github-cli (may be available in your distro) - https://cli.github.com/  - and then that becomes:

```
gh pr checkout 263
```

lucasdemarchi, 2021-04-27:
cloning it again pointing to another branch works too, although it's not necessary

highfreq, 2021-04-27:
It works great!!!!
Now can connect and disconnect and it works.
Tried disconnecting MP and even closing it. It keeps working when i reattempt to connect.
GREAT!!!!
THANK YOU VERY MUCH!!

lucasdemarchi, 2021-04-27:
@jbeyerstedt and everyone involved here, I'm going to merge that PR. Sorry for taking this long to solve it.

closed 2021-04-27 by lucasdemarchi

highfreq, 2021-04-27:
Thank you Lucas, you can't imagine how much we appreciate this patch!!

windelbouwman, 2021-04-29:
I think this issue contains two bugs, one of which is solved (the exit code upon exit with a broken connection). The other bug(the handling of the epoll socket error) is still present.

It took me some time to reproduce, but these were my steps:

- Start mavlink router
- Connect PX4 autopilot in simulation mode (not sure if this is required)
- Connect mavsdk server via TCP, and kill it after some seconds with CTRL+C. Repeat this until mavlink-routerd quits (with an error code set!). The issue is that mavlink-routerd should not quit, but remove the tcp connection, and close it.

windelbouwman, 2021-04-29:
A wrote a script to fuzz this bug:

Start mavlink router listening to tcp. Then use this script to repeatedly connect using mavsdk server:

```bash
#!/bin/bash

set -eu

APP=./path/to/build/src/mavsdk_server/src/mavsdk_server

while [ true ];
do
    echo "Starting ${APP}"
    ${APP} tcp://127.0.0.1:5760 &
    PID=$!
    echo "pid=${PID}"
    sleep 3
    echo "Killing ${PID}"
    kill -9 ${PID}
done
```

My output of mavlink router after some attempts:

```
TCP Connection [7] closed
TCP connection [7] accepted
TCP Connection [7] closed
TCP connection [7] accepted
8 messages to unknown endpoints in the last 5 seconds
TCP Connection [7] closed
TCP connection [7] accepted
poll error for fd 7
TCP Connection [7] closed
windel@PC:~/GIT/mavlink-router$ echo $?
1
```

As you can see, the exit code is properly set now, which is fixed!

jbeyerstedt, 2021-04-29:
I did some tests yesterday with my setup (MAVSDK Python exiting/ crashing without cleaning up the mavsdk instance) and had no issues any more.

I also validated beforehand, that the script still triggers the bug and used the same script after re-installing the latest mavlink-router version. As expected, I got a poll error and exit code 0 before and no poll error afterwards. It also worked multiple times and with multiple client connected.

It's fine for me now, also because a proper exit code is set now, when the poll error eventually occurs. It seems, like you have to try quite hard to get the poll error. But I don't know this epoll stuff in detail, so maybe experts on that topic have another opinion. @lucasdemarchi Should we re-open this issue?

windelbouwman, 2021-04-29:
I noticed that the poll error occurs more frequently when there is some actual traffic, like for instance a simulated autopilot connected to the mavlink-router. So to trigger an epoll error, run a simulation and then try multiple times connect / disconnect of the mavsdk_server. This will trigger the bug eventually.

It would be handy if there were a way to inject this epoll error somehow. Probably there is way to do this.

windelbouwman, 2021-04-29:
I made another change for this bug, see #264 . The reason for this change, is that I discovered that `is_valid` is not consistently set to false in case of epoll errors for tcp endpoints. Sometimes the read action comes before the epoll error, and `is_valid` is set to false. Sometimes, the epoll error occurs, and `is_valid` is still true. I found that simply ignoring the epoll error in case of tcp endpoints work fine. The next time the read function is called on the endpoint, the endpoint is marked invalid, and removed from the endpoint list.

PR #263 (2021-04-27): Do not exit on TCP disconnects
TCP connections are expected to be dynamic. Just like we accept new
connection on the fly, we should also gracefully accept disconnects.
This was the behavior before, but change when we decided to handle UART
disconnections as fatal in 7e7ea939c8d7 ("mainloop: handle epoll_wait error").
For UART, this should not be common as the OS is simply removing the
device under us due to physical disconnection. This restores the previous
behavior for TCP.

In a poll loop it's usual to handle the POLLERR before, but in this case
we leave it after the POLLIN/POLLOUT. Even if  [... cut]
commit 050a4f3d (2021-04-27): Do not exit on TCP disconnects

TCP connections are expected to be dynamic. Just like we accept new
connection on the fly, we should also gracefully accept disconnects.
This was the behavior before, but change when we decided to handle UART
disconnections as fatal in 7e7ea939c8d7 ("mainloop: handle epoll_wait error").
For UART, this should not be common as the OS is simply removing the
device under us due to physical disconnection. This restores the previous
behavior for TCP.

In a poll loop it's usual to handle the POLLERR before, but in this case
we leave it after the POLLIN/POLLOUT. Even if [... cut]