mavlink-router/mavlink-router #254: Different results between command line and configuration file.

96moustafa, 2021-03-11:
I am trying to use mavlink-router to forward mavlink messages between two PX4 simulated UAVs in the host machine and a QGC on a smartphone. 
I use the following command:
`mavlink-routerd -e 192.168.159.8:14550 127.0.0.1:14540 127.0.0.1:14541`
The smartphone ip:port is **192.168.159.8:14550** and the UAVs' udp ip:port are  **127.0.0.1:14540** and **127.0.0.1:14541** respectively.

This command works well and I can see both vehicles in QGC on my phone.

However, when I try to have the same behaviour in a config file but it does not produce the same result.
My config file is as follows:

```
[General]
TcpServerPort=0
ReportStats=false
MavlinkDialect=auto

[UdpEndpoint Drone1]
Mode = Normal
Address = 127.0.0.0
Port = 14540

[UdpEndpoint Drone2]
Mode = Normal
Address = 127.0.0.0
Port = 14541

[UdpEndpoint QGCmobile]
Mode = Normal
Address = 192.168.159.8
Port = 14550
```

I also tried to change the mode of both [UdpEndpoint Drone1] and [UdpEndpoint Drone2] to Eavesdropping but it also doesn't work.

In general, I would like to forward the mavlink messages for each drone to the other and from each drone to the QGS on the phone as well.

lucasdemarchi, 2021-03-11:
> Address = 127.0.0.0

wrong address

96moustafa, 2021-03-11:
Thanks. Sorry for such a stupid mistake. Generally, all endpoints will forward messages to each other as long as they are in normal mode, right?

lucasdemarchi, 2021-03-11:
yes, in normal mode endpoints will receive all messages that either don't have a target or are targeted to them. In eavesdrop mode, they "spy on" messages that are not for them too. In normal situations eavesdrop should only be used for debugging what's going on.

96moustafa, 2021-03-11:
> yes, in normal mode endpoints will receive all messages that either don't have a target or are targeted to them. In eavesdrop mode, they "spy on" messages that are not for them too. In normal situations eavesdrop should only be used for debugging what's going on.

Many thanks!!

closed 2021-03-11 by 96moustafa