mavlink-router/mavlink-router #257: Filtering potential based on SysID and general question

bartstech, 2021-03-31:
# #241 

In reference to issue #241 back in Nov of last year, is it possible to filter/route based on sysid of the platform? Understood that I can run more than one instance, but I was looking for a cleaner way to segment the stream. If I can control routing based on sysid to particular IP addresses, that would be ideal. If not I'll experiment in Node Red and Nifi.

Secondly, more of a dumb question, but with mavlink-router 2 out, when I did the build a week ago, did it pull ver 2? I was looking around version 2 and didn't see any particular install requirements.

jbeyerstedt, 2021-04-28:
Just to clarify the feature request: You want to have a list of System IDs on an endpoint, which specifies what messages should be sent out over that endpoint. So using the terms of network firewalls, this would be an outgoing filter.

The issue you want to solve is, that the Python MAVSDK (currently) can't handle multiple drones (system IDs) on one connection (even though the C++ version can).

bartstech, 2021-04-28:
For what I'm working towards, its not so much that MAVSDK can support
multiple vehicles, its that I also have multiple GCS's. I've read through a
lot of the posts for mavlink-router and one of the comments/issues raised
was that if more than one vehicle was on the network, then the telem etc
would feed to all stations. And so even though you are looking at vehicle 2
on your GCS, you would randomly be injected with vehicle 1's data. Since I
have several GCS's controlling one drone I wanted to see what sort of
segmentation would be possible based on criteria of the end point.
I also saw someone mention a filtering capacity for mavlink-router, but I
didn't see anything anywhere that really discusses it.
I can segment/filter the data streams by coupling in Node-Red, but I was
curious to know if it could be achieved through mavlink-router as well. So
within the config file I could establish segmentation based on sysid.
Obviously this is not something everyone is interested in, nor really
trying to do, but I'm trying to account for a potential future use case
where more than one vehicle is on the network.

jbeyerstedt, 2021-04-29:
I understand, that this might be needed in some cases and extending the filtering to a more extensive "filtering framework" is on my list of nice-to-have features. There are multiple issues requesting different filtering features. (But I simply don't have the time right now to implement all of the nice-to-have things.)

Getting the data from all vehicles on each endpoint is actually a feature of mavlink-router (documenting all the current features is another issue). And receiving data from multiple vehicles shouldn't be an issue for a ground station (e.g. QGroundControl is perfectly fine with that). So _actually_ your ground station is missing the feature to handle multiple vehicles. But this doesn't mean, that sysID filtering shouldn't be added to mavlink-router.

bartstech, 2021-04-29:
You make a good point about the GCS and the sysid. Its something I've
contemplated before but the only thing I could think of was some sort of UI
interface to select the drone, but that gets away from the core idea of not
having user interaction to switch between them.
I completely get prioritizing features. My case is not common in the least
bit.

jbeyerstedt, 2022-04-07:
Since https://github.com/mavlink-router/mavlink-router/pull/346 was merged recently, you can specify all kinds of filters on the outgoing and incoming messages (by source system and component ID, as well as message ID) on all endpoint types. All filters are allow-lists (whitelists), which is actually what you needed.

jbeyerstedt, 2022-04-07:
I'm closing this for now, if there're any more questions just add them here or open a new issue.