mavlink-router/mavlink-router #259: Connect to herelink udp stream

highfreq, 2021-04-02:
I can connect to herelink usb and activate teethering, my raspberry can ping herelink at 192.168.42.129 trough usb0 ethernet. Herelink is emitting an udp stream at port 14552.

If i start on my raspberry:
mavproxy.py --master=udpout:192.168.42.129:14552 --out=udpbcast:192.168.4.255:14550
Everything works and stream is rebroadcasted on the wifi network that the raspberry creates.

Now i would like to use mavlink-router to do the same, i pretty much tried everything but nothing seems to work:

mavlink-routerd  192.168.42.129:14552
Error binding socket (Cannot assign requested address)
Could not open 192.168.42.129:14552

mavlink-routerd -e 192.168.42.129:14552
Open UDP [4] 192.168.42.129:14552  
Error binding socket (Cannot assign requested address)
Could not open 192.168.42.129:14552

I just can't connect to source of stream no matter what i try.

jbeyerstedt, 2021-04-28:
With mavlink-router, there are two types of UDP endpoint: One listening for incoming data at a specified port ("Eavesdropping") and one for sending data to a specific IP and port ("Normal"). Using the command line syntax, the `-e` arguments are always in normal mode ("output"), while the last endpoint is created in eavesdropping mode ("input"). (The documentation of this is not clear enough.)

Your commands are trying to open an UDP port on 192.168.42.129, which isn't the IP of your host. So this has to fail, because you can only listen on the IP of any of your network devices or all devices together (using 0.0.0.0).

You're writing, that the herelink is "emitting an udp stream at port 14552", which couldn't be 100% true. The herelink can either send data to an IP and port configured at that device (from port 14552) or you can connect to it at port 14552, which seems to be the case reading you mavproxy command.
The second case isn't possible with mavlink-router, because mavlink-router would have to send a packet to the herelink first to establish a connection. I don't know mavproxy in detail, but it seems, that it'll emit heartbeats on it's own to initialize an "udpout" connection. Mavlink-router does not do this!

The only thing, you could try with mavlink-router would be this command (listening for UDP data on 14552 and sending to 192.168.4.255:14550):
```
mavlink-routerd -e 192.168.4.255:14550 0.0.0.0:14552
```

lucasdemarchi, 2021-09-04:
Hopefully the instructions above clarify. For herelink, since it's IP-based, another possibility is to skip mavlink-router altogether and just use iptables rules to route them. This is what I did for the Parrot SkyController 2. See https://github.com/ArduPilot/dema-rc/blob/master/distro/sc2/usr/bin/dema-rc-cm#L23-L33

closed 2021-09-04 by lucasdemarchi