mavlink-router/mavlink-router #261: Log mavlink communications

jaxxzer, 2021-04-08:
I do not see any option or function inside of mavlink-router to log the mavlink communication passed through the router.

There is support for dataflash logging, but I want to log all of the mavlink messages passing through the router (ie. a telemetry log).

Is this function supported?

jbeyerstedt, 2021-04-28:
There are some MavLink logging options, as you can see in the [example.conf](https://github.com/mavlink-router/mavlink-router/blob/master/examples/config.sample#L30) (not to be confused with the `DebugLogLevel` parameter, which is the application log). This should give you a ulog (PX4) or binlog (Ardupilot) file depending on the auto-detected or configured flight stack. But I personally haven't used this yet, so I can't give any other advise.

jbeyerstedt, 2021-07-28:
Closing in favor of #265

closed 2021-07-28 by jbeyerstedt