mavlink-router/mavlink-router #265: Add support to create Telemetry Logs (tlog) files

patrickelectric, 2021-04-30:
https://ardupilot.org/copter/docs/common-mission-planner-telemetry-logs.html
This is a common log file structure for analysis software.

jbeyerstedt, 2021-05-26:
I don't really know Ardupilot, but from what I've read in the docs you linked, I haven't understood the difference between the bin-log format and tlog. (Other than, that a log written by the FMU might have more/ other information, than what was transmitted via MAVLink.)

So asking a bit provocative: Why doesn't the bin-log format work for you? Mavlink-router should auto-detect the used flight stack and chose the format accordingly (see the [Readme](https://github.com/mavlink-router/mavlink-router#flight-stack-logging)).

lucasdemarchi, 2021-05-26:
@jbeyerstedt telemetry log, or tlog, would be a dump of all the mavlink messages traversing that endpoint. You may want not to use binlog if your communication with the drone has limited bandwidth. In that case you are probably configuring the binlog to be saved locally in the drone to an external storage like microsd card, but it would still be nice to have all the communication with the GCS recorded.

All the GCSs I know of already  have the capability to record those, but IMO it's not a bad idea to have support for it in mavlink-router too.

jbeyerstedt, 2021-05-27:
I think, we misunderstood each other. It's clear for me, why I want to record logs to the SD card on the FMU. But what would be the difference between mavlink-router writing a binlog of the messages and a tlog? Is the binlog (and ulog) one file per system ID and the tlog would just be a dump of everything? Because regardless of the file format, it wouldn't have more information (set of MAVlink messages at their data rate) than what's sent by the FMU. So the content would be the same for all file formats, isn't it?

I'm definitely not against adding this feature, I'm just curious to understand, where the difference is.

lucasdemarchi, 2021-06-01:
@jbeyerstedt there are reasons not to stream the binlog over mavlink. Notice that in this case mavlink is the transport for another format. On pretty good reason for not enabling binlog (or ulog if talking about PX4) is that it takes up considerable bandwidth on that link. Since you are more familiar with PX4, back in the time when I was working with it, any baudrate below 384k would cause dropped packets and incomplete log.

It's also not negligible on the flight controller side the additional delay introduced for handling those messages.

So, I think that having an option to save the tlog is useful for those not enabling the binlog/ulog stream over mavlink.

jbeyerstedt, 2021-06-01:
OK, thanks. I wasn't aware, that the current log implementation is actually requesting another data stream from the FMU. Somehow I expected, that mavlink-router is just logging the messages, it receives anyways. (At a second thought, it obvious, because the ulog is actually logging uORB messages instead of MAVLink.)

So the requested tlog feature is actually what I expected to be be implemented already. Thanks for getting this straight in my head.

patrickelectric, 2021-06-04:
Check https://github.com/mavlink-router/mavlink-router/pull/269, I would like for some feedback about how to deal with tlog in the command line interface and what should be the best way to accomplish it with ulog and binlog.

closed 2023-11-22 by patrickelectric