mavlink-router/mavlink-router #267: build issue

livip2006, 2021-05-18:
"./autogen.sh &&  ./configure ... --prefix=/usr"on raspberry 4B show "checking whether the C compiler works ..no configure: error: C
 compiler cannot create executables
the config.log shows "gcc:error ;unrecognized command Line option ‘-V’ ，"gcc:error:unrecognized command line option '-02'

lucasdemarchi, 2021-05-18:
what distro? should be sufficient to check the docs for that distro on how to properly install compiler and dev tools

livip2006, 2021-05-18:
> 
> 
> what distro? should be sufficient to check the docs for that distro on how to properly install compiler and dev tools

distro of master,but cannot find docs for compiler and dev tools

lucasdemarchi, 2021-05-18:
I meant "what linux distribution?". Examples: ubuntu, fedora, archlinux, debian, raspbian, etc

livip2006, 2021-05-18:
> 
> 
> I meant "what linux distribution?". Examples: ubuntu, fedora, archlinux, debian, raspbian, etc

latest raspbian (raspberry PI osRelease date: March 4th 2021 Kernel version: 5.10)

livip2006, 2021-05-18:
> 
> 
> > I meant "what linux distribution?". Examples: ubuntu, fedora, archlinux, debian, raspbian, etc
> 
> latest raspbian (raspberry PI osRelease date: March 4th 2021 Kernel version: 5.10)

and the complier is gcc 8.3.0

lucasdemarchi, 2021-05-18:
I don't use it, but googling it says `sudo apt-get install build-essential` should bring what you need

livip2006, 2021-05-18:
> 
> 
> I don't use it, but googling it says `sudo apt-get install build-essential` should bring what you need
I already try this,the newest version gcc does not support build and shows errors,And I followed to downgrade the version of gcc,but failed

jbeyerstedt, 2021-05-27:
I'm running Raspbian (earlier image, but with the latest updates) on a Raspberry Pi 3 B and it just works fine.

```
$ gcc --version
gcc (Raspbian 8.3.0-6+rpi1) 8.3.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

$ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 10 (buster)
Release:	10
Codename:	buster

$ uname -a
Linux <redacted> 5.10.17-v7+ #1403 SMP Mon Feb 22 11:29:51 GMT 2021 armv7l GNU/Linux
```

IIRC, I just installed build-essential as mentioned by Lucas and nothing else compiler related on my system. Could you try to re-install the build-essential package? Either just with `sudo apt reinstall build-essential` or even the "hard" way:
```shell
sudo apt remove build-essential
sudo apt autoremove
sudo reboot
sudo apt install build-essential
```

Posting a link to the config.log and command line output (e.g. at gist.github.com) might also help our guessing, where the issue is.

lucasdemarchi, 2021-09-04:
Since Raspbian is debian-based, it likely needs the same set of packages. See #279

closed 2021-09-04 by lucasdemarchi