mavlink-router/mavlink-router #272: read me description for routing is misleading

akkawimo, 2021-07-02:
"mavlink-router mainly routes mavlink packets from one endpoint to the other endpoints without differentiating what they are."

- Messages will be processed based on the mavlink routing standard. https://mavlink.io/en/guide/routing.html#routing-detail  
-There is a vector for each endpoint instance that adds new sender sysid/compid pairs that it sees. If a message with sender sys_id and compid is received that is listed in the endpoint vector, it will be rejected.

lucasdemarchi, 2021-07-05:
can you point can you detail what exactly is misleading or maybe propose a change to the README.md through  a patch?

akkawimo, 2021-07-06:
"mavlink-router mainly routes mavlink packets from one endpoint to the other endpoints without differentiating what they are."

without differentiating what they [mavlink packets] are.

If I'm not mistaken, this is not the case. There is some filtering happening which will appear as dropped packets if you assume that the ports are just copies of one another.

jbeyerstedt, 2021-07-06:
My understanding of that sentence is, that packets are routed between endpoint without differentiating between the endpoints. So the "they" is referencing to the last noun.

akkawimo, 2021-07-06:
yea that makes sense. Okay, maybe still add something about the filtering?

commit 5518d4cc (2021-09-04): README: clarify how endpoints can be differentiated

Fix: #272
closed 2021-09-10 by lucasdemarchi