mavlink-router/mavlink-router #273: high routing latency when FC is connected via usb

akkawimo, 2021-07-02:
Setup:
- Tested on v2 mavlink router
- Device=/dev/ttyACM0
- Baud=115200

Routing:
- QGC > MR over udp
- mavlink handler on Companion > MR on tcp
- FC > MR on usb

Screened ports using tcpdump:
- After disconnecting QGC. QGC heartbeat in the udp stream stops but keeps showing up on tcp.
- Latency varies but at least a few minutes depending on how long it's been running.

jbeyerstedt, 2021-07-28:
I can only guess what's going wrong here, but it sounds like there's some re-transmission happening on the TCP link. Either because some packets were lost on the way to the companion or the TCP-ACK was lost somehow. I don't know, at which intervals TCP tries to re-transmit lost packets. But if some kind of exponential backoff is used, the re-transmission "echo" can last some time.

Could you try to run mavlink-router in a shell (instead of via systemd) with the `-v` verbose flag and check, if the QGC heartbeat is still being transmitted via the TCP companion endpoint after you have disconnected QGC? If it is showing up there, my theory above isn't true and there must be another reason. Otherwise it's in the lower layers of TCP itself and TCP just behaves as it's intended to work.

thealienthing, 2021-08-24:
Any progress on this issue? I'm running a similar build with USB connection to flight controller, but I'm running a UDP connection between MR and Mission Planner and everything appears to be working. I'll try a TCP connection to see if I get similar behavior.

lucasdemarchi, 2021-09-05:
@akkawimo  not sure I understand the setup, sorry. What's "mavlink handler on companion > MR on tcp"? Do you have another sw receiving the mavlink stream and forwarding it to mavlink-router? Is this TCP connection on the same board, using localhost or a different one?

lucasdemarchi, 2022-08-04:
@akkawimo  any update?