mavlink-router/mavlink-router #274: OpenWRT package for raspberry

highfreq, 2021-07-14:
Any chance to have some guidance on how to have mavlink router working on a raspberry that runs OpenWRT? (actually openmptcprouter)
thanks in advance.

highfreq, 2021-07-23:
I nstalled everything to try to cross compile it to openwrt on raspberry and run the sample, it compiled and run on openwrt. Now apply all this to cross compile mavlink-router is another story. Anybody willing to help a bit?

vooon, 2021-07-26:
That should be relatively easy, here some examples that you could use:

https://github.com/openwrt/packages/blob/master/net/ser2net/Makefile
https://github.com/openwrt/packages/blob/master/utils/avrdude/Makefile

Only problem that you need to package mavlink as well.

closed 2021-09-04 by lucasdemarchi

booo, 2022-05-09:
I successfully build mavlink-router with the following Makefile:

```make
include $(TOPDIR)/rules.mk

PKG_NAME:=mavlinkrouter
PKG_RELEASE:=0.0.0

PKG_SOURCE_PROTO:=git
PKG_SOURCE_URL=https://github.com/mavlink-router/mavlink-router
PKG_SOURCE_VERSION:=master

PKG_BUILD_DIR=$(BUILD_DIR)/$(PKG_NAME)/$(PKG_NAME)-$(PKG_VERSION)

# do we need this?
PKG_INSTALL:=1

include $(INCLUDE_DIR)/package.mk
include $(INCLUDE_DIR)/meson.mk

MESON_ARGS+=-Dsystemdsystemunitdir=/usr/lib/systemd/system

define Package/mavlinkrouter
 SECTION:=net
 CATEGORY:=Network
 TITLE:=mavlinkrouter
 DEPENDS:=+libstdcpp
endef

define Package/mavlinkrouter/description
	Route mavlink packets between endpoints
endef

define Package/mavlinkrouter/install
	$(INSTALL_DIR) $(1)/usr/bin
	$(CP) $(PKG_INSTALL_DIR)/usr/bin/mavlink-routerd $(1)/usr/bin
endef

$(eval $(call BuildPackage,mavlinkrouter))
```

I still have to test this carefully on hardware.