mavlink-router/mavlink-router #275: Error connecting to socket (Connection refused)

paravarzar-shahrokh, 2021-07-28:
Hi, 
I would like to route the packets from a serial to local 127.0.0.1:2000 but I am receiving the following error
`Error connecting to socket (Connection refused)
Could not open 127.0.0.1:2000.
TCP Connection [4] closed
Open UDP [5] 127.0.0.1:3000  
Error while trying to write serial port latency: Unknown error -1
Open UART [6] /dev/ttyTHS0 *
UART [6] speed = 57600
Open TCP [7] 0.0.0.0:5760 *
Error connecting to socket (Connection refused)
Error connecting to socket (Connection refused)
Error connecting to socket (Connection refused)
Error connecting to socket (Connection refused)
Error connecting to socket (Connection refused)
Error connecting to socket (Connection refused)
Error connecting to socket (Connection refused)
Error connecting to socket (Connection refused)
`
here is my config file
`[UartEndpoint Rudi]
Device = /dev/ttyTHS0
Baud = 57600

[TcpEndpoint local]
Address = 127.0.0.1
port = 2000
RetryTimeout = 10

[UdpEndpoint local2]
Mode = Normal
Address = 127.0.0.1
port = 3000
`
and here is the ufw output 
`Status: active
Logging: on (low)
Default: deny (incoming), allow (outgoing), disabled (routed)
New profiles: skip

To                         Action      From
--                         ------      ----
1:9999/tcp                 ALLOW IN    Anywhere                  
22                         ALLOW IN    Anywhere                  
3000                       ALLOW IN    Anywhere                  
3000/tcp                   ALLOW IN    Anywhere                  
2000/tcp                   ALLOW IN    Anywhere                  
1:9999/tcp (v6)            ALLOW IN    Anywhere (v6)             
22 (v6)                    ALLOW IN    Anywhere (v6)             
3000 (v6)                  ALLOW IN    Anywhere (v6)             
3000/tcp (v6)              ALLOW IN    Anywhere (v6)             
2000/tcp (v6)              ALLOW IN    Anywhere (v6)`

jbeyerstedt, 2021-07-28:
It seems, like nothing is currently listening on port 2000 on localhost, so the connection fails. This shouldn't be an issue of with your firewall by the way, because everything is happening on the loopback interface/ localhost.

But do you really want to use TCP to forward the mavlink messages? Normally UDP should be enough. What are you using to receive the messages on port 2000?

paravarzar-shahrokh, 2021-07-28:
I am testing the stability of connection, we are going to pass the packets to another software using the TCP/UDP in a specific port. I want to use the TCP. Should something necessarily listen to this port?

jbeyerstedt, 2021-07-28:
> Should something necessarily listen to this port?

Yes, that's how a TCP connection works. Somebody has to be the server (listening on a certain port for new client connections) and another one can be the client connecting to that server.

Mavlink-router has two options to initiate a TCP link between two applications:
1. mavlink-router is the server (see the `TcpServerPort` parameter in the `[general]` section of the config file) and waits for some application to connect. As long as a client is connected to that TCP port (default value: 5760), it'll receive data from all other endpoints and can send data to all other endpoints.
2. You can add a TcpEndpoint, which will act as a client and connects to another "server" application.

As I understood your plan and config file, you tried to do the second option. Then your analysis application must already run to let mavlink-router connect to it. Otherwise, as you can see in the log, mavlink-router will re-try to reach your TCP server at the configured interval and no data is transmitted to that endpoint until a connection was made.

On the other hand, the UdpEndpoint in Normal mode will just send packets to the specified IP and port without caring, if the data is received by somebody. (Thats how UDP works: No client-server sessions and no check if anybody has received the data.)

paravarzar-shahrokh, 2021-07-28:
So if I want to see if the mavlink router is sending the packages to other application and the other application are being fed by the mavlink-router ( the other application is just listening and using the mavlink router packages) then I should define the TcpEndpoint such 127.0.0.1:3001? and then use the wireshark to check the packages and stability of the both the mavlin-router and XXXX software? 
Mavlink-router ----------> application ? if application receiver the data ?

jbeyerstedt, 2021-07-28:
Your config is fine. You just have to start something listening on 127.0.0.1, TCP port 2000. Then you can use Wireshark to record and analyze the TCP data stream.

paravarzar-shahrokh, 2021-07-28:
Ok Great, Thanks for supporting me.

closed 2021-07-28 by paravarzar-shahrokh