mavlink-router/mavlink-router #282: Log to file error when another mavlink object present

highfreq, 2021-09-16:
I get errors in log writing on file whenever i have any other mavlink object on the aircraft (in my case basecam).
I have had this problem for years now, if i enable log to file and store it with mavlink-router it works if ardupilot is the only object that has a mavlink id on board. As soon as i have any other object with a mavlink id it generates an error and doesn’t write logs to companion directory anymore. As soon as i remove the gimbal it restart working again.
All id are unique, no duplicated. Actually gimbal is on another system id, before i moved it to another system id it was same id as ardupilot but different component. In both situations the problem in logging was still there.
I tried both auto dialect and forcing ardupilotmega, both give same error.

Any help would be greatly appreciated.

lucasdemarchi, 2021-09-16:
I will try to reproduce it locally. Probably easier to start with the simulator running and another script as the third system.

highfreq, 2021-09-16:
Would be great if it can be solved. If there is anything i can do to help please ask.

highfreq, 2021-09-17:
If the problem doesn't show up maybe you need to connect phisically to 2 different serial ports on the controller. In my setup companion and gimbal are connected to serial 1 and serial 2.

highfreq, 2021-09-24:
Hello Lucas, did you have a chance to check the error i mentioned?

Thanks

highfreq, 2021-10-05:
Hello Lucas, is there any news by any chance? Did you replicate the problem?

lucasdemarchi, 2021-10-06:
I think I forgot to answer here, sorry. I tried running it, but couldn't reproduce. Some questions:
- Are you running Ardupilot and the mavlink packets being routed from the gimbal to the uart going to the companion computer where mavlink-router is running?
- What are the system and component IDs you are using for each of them?

Yesterday I was preparing some commands so we could get a .pcap to be created and analyzed with wireshark. The easiest way would be to create an additional endpoint in eavesdrop mode. However yesterday I noticed the eavesdrop mode is completely broken and will need to be fixed.  I will take a look if I can fix it this week.

lucasdemarchi, 2021-10-16:
I tried to reproduce again, this time adding another component to ardupilot. I couldn't reproduce it in sitl:

```
./build/sitl/bin/arducopter  --defaults copter.parm --model x --uartA udpclient:127.0.0.1:5000 --uartC udpclient:127.0.0.1:5002
```

params LOG_* already set appropriately to log via mavlink.

I used ./examples/heartbeat-sender.py from https://github.com/mavlink-router/mavlink-router/pull/285 - it seems to work fine.

highfreq, 2021-10-16:
Ok than i guess it works fine.

highfreq, 2021-11-07:
@lucasdemarchi Don't know if it can be of any help but i just reinstalled mavlink-router from source and i get a bunch of 55,8kb files between flights.

lucasdemarchi, 2021-11-07:
I reproduced the issue with log files restarted. The restarting logic is broken when ardupilot is configured with LOG_DISARMED=0. Because then we don't know what happened since it simply stops responding.

lucasdemarchi, 2021-11-07:
Btw this other issue you are describing is probably #221

highfreq, 2021-11-07:
Yes sorry same, didn't find that other post. Is there anything in the works to solve?

lucasdemarchi, 2021-11-07:
Not yet, sorry. I'm doing some major changes and hopefully will get to rewrite that part soon

highfreq, 2021-11-07:
great thanks!!!!

highfreq, 2021-11-22:
Hello Lucas, noticed all the changes in mavlink-router, in the actual version is there any change to logging?

thanks

highfreq, 2021-12-02:
Hello Lucas, is there any news on this issue by any chance?

highfreq, 2022-03-13:
@lucasdemarchi at least describe on home that log is broken so people don't loose their time using it.

Katawann, 2022-08-17:
Hello,

I don't get exactly what the issue is as there is no description, but just a quick note to mention that on my side the logging option is working fine 

Setup: Pixhawk4 with PX4, Raspberry connected through USB to Pixhawk to get the log
main.conf: LogMode = while-armed
command: mavlink-routerd /dev/ttyACM0:926100