mavlink-router/mavlink-router #295: How to get "git submodule update --init --recursive" using https instead of ssh

highfreq, 2021-11-01:
I am running "git submodule update --init --recursive" and get a permission error. I understand i should use https instead of ssh.
Anyone could please let me know how to use https during "git submodule update --init --recursive"

Thanks.

highfreq, 2021-11-01:
Answer my question for others that may have same problem.
On my raspberry i had to do the following:

sudo nano /home/pi/mavlink-router/.git/config

change submodule url with the following:

url = https://github.com/mavlink/c_library_v2.git

save and exit.

My question is why all of a sudden i have to do this while untill 2-3 days ago i could use ssh?

lucasdemarchi, 2021-11-01:
@highfreq this was fixed in #293 - I inadvertently changed it to ssh when I the submodule changed in #291

highfreq, 2021-11-01:
Yes, seen it, thanks.