mavlink-router/mavlink-router #307: Improve setup documentation

nirgivon, 2021-12-07:
Hello,

I am not a Linux user, but I can find my way around.

I found the setup documentation very difficult to work with.

for example:
It is not clear that the first step is cloning the repo.
/etc/mavlink-router/main.conf is not created by default - it is not clear
MAVLINK_ROUTER_CONF_DIR is not created by default - it is not clear

after build/install:
How to run mavlink-routed?
Dose it start atomicity after reboot?
Where is the executable file?

Should a config file be created before running?
What happens if running without a config file?

I was able to build and install successfully, but unable/don't know how to run the service.
very frustration.

jbeyerstedt, 2021-12-07:
The Readme states regarding the configuration (file):
> There are two ways to configure mavlink-router which can be used in combination: Configuration file(s) and command line parameters.

I thought, that "_can_ be used in combination" clearly specifies, that you can either have just a config file, just the CLI parameters or both in combination. But I can add another sentence in my open PR, which adds even more details on the routing behavior (#305).

Reading compiling, installation and running: I thinks it's fair to assume some basic knowledge of Linux when using this software. The readme has a phrase, that a systemd unit will be installed (which answers the question of running it on reboot). The binary is installed in the standard location (which should be in your $PATH), but can be changed (see: "Installation location can be changed using the -Dprefix option to meson setup.")

nirgivon, 2021-12-07:
Hi,
Thanks for the very very fast replay :-)

I do have basic knowledge of Linux, If I did not, I was not able to complete the installation.
But IMHO, system is beyond basic knowledge.

Now, I am sorry if it sounded like a criticism, but this is not the case.

Please understand this:
I am a user of this project, looking at the SSH of my pi after the installation.
And I have no idea what to do.
dose it work? how do I run it?
if I create a config file, do i need to reboot the pi?

this is not criticism, this is feedback from the factory flor of a user that don't understand how to use.

jbeyerstedt, 2021-12-07:
OK, so I would add the words "system administration" to linux knowledge. I don't think, that a systemd how-to should be part of the Readme though, because it this information should be easily accessible in the internet.

So after having built and installed the software, there should be a systemd unit called `mavlink-router`, which uses the configuration file. You can start it using `systemd start mavlink-router` and use the provided python scripts to pass some test messages through it. But you wouldn't see many things in the log, because it's quite minimal on purpose.

The better way (in my opinion) to check, that mavlink-router is doing something, is to manually start mavlink-router. You should be able to type the commands from the [Readme's CLI Parameters](https://github.com/mavlink-router/mavlink-router#cli-parameters) section in your shell and run the application this way. Adding `-v` will output some lines for each message.

nirgivon, 2021-12-07:
Thanks again for the fast replay!

Now that helped me a lot!

jbeyerstedt, 2022-01-17:
fixed by #327

closed 2022-01-17 by jbeyerstedt