mavlink-router/mavlink-router #311: Can the 56k files factory be stopped please?

highfreq, 2021-12-20:
Just compiled this morning latest patch to logging and the log is a joke. It produces a 56k file every 11 seconds when not armed and it just keeps going. If i arm it logs correctly and as soon as i disarm it restarts to make 56k files, over and over.
This thing has been going on for months now, can it be fixed please?

lucasdemarchi, 2021-12-20:
@highfreq if you can't provide a patch, it would be good to tone down. You also have the possibility to pay someone to submit the fix.

I will close this issue here as there is already one opened for this: #221 with more details on #282.

closed 2021-12-20 by lucasdemarchi

highfreq, 2021-12-20:
Tone doesn't look too bad to me. Other than that maybe you should have left it opened so that when someone installs it and mavlink-router spams his companion with hundreds of 56k files, at least, he knows why.