mavlink-router/mavlink-router #312: Can the 56k files factory be stopped please?

highfreq, 2021-12-20:
Just compiled this morning latest patch to logging and the log is a joke. It produces a 56k file every 11 seconds when not armed and it just keeps going. If i arm it logs correctly and as soon as i disarm it restarts to make 56k files, over and over.
This thing has been going on for months now, can it be fixed please?

jbeyerstedt, 2021-12-22:
Is this the same issue as #221?

jbeyerstedt, 2021-12-22:
As stated in #311, this is not a good issue description **and** it's a duplicated issue. So re-posting it doesn't make anything better. This project isn't directly funded by anybody. As far as I know, many contributors are working on it in their spare time. There might be some like me, which use mavlink-router at work and therefore can do some fixes and additions in their working hours. But then, I can only justify to fix things we need. What I want to say is, that ranting and spamming us with duplicated issues, doesn't increase the willingness to fix it.

I would like to close this issue, because it's quite useless for:
- the developer community because the issue was already reported with a detailed description
- any user because it doesn't hit the most obvious search/ the title isn't very useful (I would search for "many log files" instead of "56k files factory" when I have that issue)

highfreq, 2021-12-22:
Don't worry i fixed the issue myself, took me around 3 hours yesterday afternoon. 

Can close it, thanks.

jbeyerstedt, 2021-12-22:
Feel free to open a PR so we can add this fix in the next release, which is planned for January.

closed 2021-12-22 by jbeyerstedt

highfreq, 2021-12-22:
Will do when i'll have some spare time, it works for me now, so i am ok.