mavlink-router/mavlink-router #320: Installation process

kickfliph, 2022-01-01:
Hello there,

HAPPY NEW YEAR to everyone. 

I'm trying to install the mavlink-router on a Raspberry Zero 2 W, at least for me the compilation takes around 10 hours, I run through many problems and errors, and in the end, was not successfully installed. I follow the instruction installation process on the git page. The count down "[1/23]" of the compilation packages went to show some errors compiling some of the packages and then stoping, what I did was just re-run the command "ninja -C build" and from what I see the installation cat up from the last package successfully compile and the count down continues to grows. In the end, the binary is installed on my /usr/bin directory but I can't see the /etc/mavlink-router/main.conf file installed, if I run the binary from the console I'm getting this message:
 
$ mavlink-routerd
mavlink-router version v2-145-gdd4de3c
Could not open conf file '/etc/mavlink-router/main.conf' (No such file or directory)
Opened TCP Server [4] 0.0.0.0:5760

Looks like is running, but I really don't know if it's fully operational. Please share with me any of your thoughts I really appreciate all your help.

Regards,

khancyr, 2022-01-01:
Hello,
To install you should run ninja -C build install, without install you just build the binaries

kickfliph, 2022-01-01:
Hello @khancyr,

Thanks for reaching back. So to get this straight, these are the steps I'm following:

1.- sudo apt install git meson ninja-build pkg-config gcc g++ systems and cd into the folder.
2.-  meson setup build .
3.- ninja -C build
and then
4.- sudo ninja -C build install

So you saying that now I should remove the build folder that has been created by running the number 2 step and just run number 4 instead and this will give me the expected result? 

Regards.

lucasdemarchi, 2022-01-03:
I would really recommend people to cross-compile when doing it for targets that are not powerful enough

kickfliph, 2022-01-03:
Thanks, @lucasdemarchi,

I really appreciate your reaching back. Is there any documentation I can follow in order to do that?

Regards,

jbeyerstedt, 2022-01-06:
Normally, I would recommend using https://github.com/dockcross/dockcross, but that doesn't support `meson` unfortunately.

Otherwise looking at our [CI pipeline](https://github.com/mavlink-router/mavlink-router/blob/master/.github/workflows/main.yml), something like this should be a starting point:
```
sudo apt-get install --yes ninja-build gcc-arm-linux-gnueabihf g++-arm-linux-gnueabihf podman python3-pip
pip3 install meson pymavlink

mkdir rootfs
tools/ci-create-rootfs.sh armhf > rootfs.tar
tar -xf rootfs.tar --directory=rootfs
sed -e "s#@sysroot@#$PWD/rootfs#g" meson-cross-sysroot.ini.in > meson-cross-sysroot.ini

meson setup -Dsystemdsystemunitdir=/usr/lib/systemd/system --cross-file meson-cross-armhf.ini --cross-file meson-cross-sysroot.ini build-armhf .

ninja -C build-armhf

DESTDIR=./.debpkg ninja -C build-armhf install
```
Run the steps above on your normal PC. I'm not sure about the installation step though.

lucasdemarchi, 2022-01-06:
Yeah... Something like that would work. Just the rootfs part that probably needs adapting. If you have the board around, you can just mount it via e.g. sshfs. Whatever distro you use in your rpi (e.g. Raspbian, RaspberryPi OS, Ubuntu, fedora, etc) should match what you are using as rootfs to build it.

lucasdemarchi, 2022-01-13:
Just tested with a RPI CM4 and commands above are mostly corrected. I don't use ubuntu so the host setup for me was a little different. But for the rest you may want to replace these lines:

```
tools/ci-create-rootfs.sh armhf > rootfs.tar
tar -xf rootfs.tar --directory=rootfs
```

with:

```
mkdir rootfs
sshfs pi@192.168.2.43:/ rootfs/
```

Another small detail if you are using gtest is that Raspberrypi OS uses  /usr/lib/arm-linux-gnueabihf/pkgconfig instead of /usr/lib/pkgconfig, so you will have to tweak that as well.

xiew, 2022-05-31:
Hey all, so it seems like it'g got something to do with the Raspberry Pi Zero 2 or the way meson is building Mavlink router on the Zero 2. It compiles fine on other Pis. Have tried on the Pi 4 and Pi 3B+. Here is where the build fails:

```
ninja: Entering directory `build'
[11/24] Compiling C++ object src/mavlink-routerd.p/binlog.cpp.o
FAILED: src/mavlink-routerd.p/binlog.cpp.o
c++ -Isrc/mavlink-routerd.p -Isrc -I../src -I../modules/mavlink_c_library_v2/ardupilotmega -I. -fdiagnostics-color=always -pipe -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wnon-virtual-dtor -Wextra -std=gnu++11 -O2 -g -Wno-inline -Wundef -Wformat=2 -Wlogical-op -Wsign-compare -Wformat-security -Wmissing-include-dirs -Wformat-nonliteral -Wpointer-arith -Winit-self -Wfloat-equal -Wredundant-decls -Wmissing-declarations -Wmissing-noreturn -Wshadow -Wendif-labels -Wstrict-aliasing=3 -Wwrite-strings -Wno-long-long -Wno-overlength-strings -Wno-unused-parameter -Wno-missing-field-initializers -Wno-unused-result -Wchar-subscripts -Wtype-limits -Wuninitialized -include config.h -pthread -MD -MQ src/mavlink-routerd.p/binlog.cpp.o -MF src/mavlink-routerd.p/binlog.cpp.o.d -o src/mavlink-routerd.p/binlog.cpp.o -c ../src/binlog.cpp
{standard input}: Assembler messages:
{standard input}: Warning: end of file not at end of a line; newline inserted
{standard input}:417: Error: unknown pseudo-op: `.lbe2'
{standard input}: Error: open CFI at the end of file; missing .cfi_endproc directive
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
[16/24] Compiling C++ object src/mavlink-routerd.p/endpoint.cpp.o
ninja: build stopped: subcommand failed.
```

meson setup build . gives this:

```
The Meson build system
Version: 0.56.1
Source dir: /home/sciii/mavlink-router
Build dir: /home/sciii/mavlink-router/build
Build type: native build
Project name: mavlink-router
Project version: 2
C compiler for the host machine: cc (gcc 8.3.0 "cc (Raspbian 8.3.0-6+rpi1) 8.3.0")
C linker for the host machine: cc ld.bfd 2.31.1
C++ compiler for the host machine: c++ (gcc 8.3.0 "c++ (Raspbian 8.3.0-6+rpi1) 8.3.0")
C++ linker for the host machine: c++ ld.bfd 2.31.1
Host machine cpu family: arm
Host machine cpu: armv7l
Library m found: YES
Library rt found: YES
Run-time dependency threads found: YES
Found pkg-config: /usr/bin/pkg-config (0.29)
Run-time dependency systemd found: YES 241
Run-time dependency GTest found: NO (tried pkgconfig and system)
Compiler for C supports arguments -Wno-inline: YES
Compiler for C supports arguments -Wundef: YES
Compiler for C supports arguments -Wformat=2: YES
Compiler for C supports arguments -Wlogical-op: YES
Compiler for C supports arguments -Wsign-compare: YES
Compiler for C supports arguments -Wformat-security: YES
Compiler for C supports arguments -Wmissing-include-dirs: YES
Compiler for C supports arguments -Wformat-nonliteral: YES
Compiler for C supports arguments -Wold-style-definition: YES
Compiler for C supports arguments -Wpointer-arith: YES
Compiler for C supports arguments -Winit-self: YES
Compiler for C supports arguments -Wdeclaration-after-statement: YES
Compiler for C supports arguments -Wfloat-equal: YES
Compiler for C supports arguments -Wmissing-prototypes: YES
Compiler for C supports arguments -Wstrict-prototypes: YES
Compiler for C supports arguments -Wredundant-decls: YES
Compiler for C supports arguments -Wmissing-declarations: YES
Compiler for C supports arguments -Wmissing-noreturn: YES
Compiler for C supports arguments -Wshadow: YES
Compiler for C supports arguments -Wendif-labels: YES
Compiler for C supports arguments -Wstrict-aliasing=3: YES
Compiler for C supports arguments -Wwrite-strings: YES
Compiler for C supports arguments -Wno-long-long: YES
Compiler for C supports arguments -Wno-overlength-strings: YES
Compiler for C supports arguments -Wno-unused-parameter: YES
Compiler for C supports arguments -Wno-missing-field-initializers: YES
Compiler for C supports arguments -Wno-unused-result: YES
Compiler for C supports arguments -Wnested-externs: YES
Compiler for C supports arguments -Wchar-subscripts: YES
Compiler for C supports arguments -Wtype-limits: YES
Compiler for C supports arguments -Wuninitialized: YES
[... 23 lines cut ...]
Compiler for C supports arguments -Wno-unused-result: YES (cached)
Compiler for C supports arguments -Wchar-subscripts: YES (cached)
Compiler for C supports arguments -Wtype-limits: YES (cached)
Compiler for C supports arguments -Wuninitialized: YES (cached)
Compiler for C++ supports arguments -Waddress-of-packed-member: NO
Header <aio.h> has symbol "aio_init" : YES
Configuring config.h using configuration
Configuring mavlink-router.service using configuration
Program run-clang-tidy found: NO
Build targets in project: 6

Found ninja-1.10.1 at /usr/bin/ninja
```
Funnily enough, if you run ninja -C build again it pulls through and you can go ahead and install it on the system. 

The functionality is eroded though - when reading serial it is not stable and the dreaded:

```
poll error for fd XX
Critical fd XX got error, exiting
```

is shown. When the connection does settle down, it eventually disconnects after a short while, throwing the poll error above...

Wonder if it has something to do with the pi zero 2. Power wise, it is a quad core 1GHz device so I thought it might be fine for build on device. Hope this helps with narrowing down where the build fails.

eli-schwartz, 2022-06-01:
Does it help to run `ninja -j1` to ensure that ninja only compiles a single file at a time, potentially using much less CPU/memory? By default ninja tries to use the equivalent of `make -J$(nproc)`. Doing the slow-but-steady route might be desirable in your case.

xiew, 2022-06-03:
Thanks eli! Yes been meaning to try this but didn't manage to get down to it.
Turns out it works flawlessly and the build compiles in a few minutes without a snag. Tested on a Raspberry Pi Zero 2.

For anyone using a Raspberry Pi Zero or equivalent, the steps I used that worked are:

```
git clone https://github.com/mavlink-router/mavlink-router.git

cd mavlink-router

git submodule update --init --recursive

sudo apt install git meson ninja-build pkg-config gcc g++ systemd build-essential

meson setup build .

sudo ninja -j1 -C build

sudo ninja -C build install
```

Then,

sudo mkdir /etc/mavlink-router

then add your main.conf file into /etc/mavlink-router

fire her up and you're ready to go!

This question/issue can be closed (=

closed 2022-08-04 by lucasdemarchi