mavlink-router/mavlink-router #324: Message routing: Bug in `has_sys_id` ?

jbeyerstedt, 2022-01-12:
While writing unit tests for the endpoints, I stumbled over this line in `Endpoint::has_sys_id()`:
https://github.com/mavlink-router/mavlink-router/blob/85d2a77fb93c2d54b10fcc09369eafa3464b057d/src/endpoint.cpp#L418

In my understanding, the condition should only be true, if the `sysid` is in the list of `_sys_comp_ids`.

But this line is always true since the bitwise OR will set all bits of `sysid` and then compare with itself!?

jbeyerstedt, 2022-01-12:
See https://github.com/mavlink-router/mavlink-router/commit/0931d6c28bf166e0342e09a0b16400e073801c30 for a fix (when I'm right with my assumption about the intended behavior). I'll add this to PR #305 after finalizing the discussion on the spec.

lucasdemarchi, 2022-01-12:
Looks like a bug indeed. should be `(id >> 8) == (sysid & 0xff)`

jbeyerstedt, 2022-01-17:
fixed by #328

closed 2022-01-17 by jbeyerstedt