mavlink-router/mavlink-router #342: Feature request: Depend on network in systemd unit

jbeyerstedt, 2022-02-17:
We had some issues with mavlink-router not starting reliable at boot, especially when a UART connection is used. It seems, like systemd starts mavlink-router too early, because it's systemd unit specifies no dependencies at all.

Adding `After=network.target` to the systemd unit fixed all reliability issue. This might not be an actual fix, but more of a work around. But it also seems reasonable to depend on networking, if most of the use cases for mavlink-router involve routing from or to network connections. (I would assume, that just connecting UART devices on a device without network at all is a quite extreme edge case.)
Is there any special reason, that the systemd unit does not include an `After=network.target` directive?

lucasdemarchi, 2022-02-17:
No special reason, other than there is no easy fix. When used in a companion computer, mavlink-router may be used to route packets only to the localhost. Also "After=network.target" doesn't really fix it. See https://www.freedesktop.org/wiki/Software/systemd/NetworkTarget/

If you want to listen on a specific address, you'd need to order after network-online.target. But just like the mavlink-router configuration, that is highly dependent on the user configuration.  It's not too bad that systemd supports override snippets, so you may just add what you need for your specific configuration.

Another solution as pointed out in that page is to use rtnetlink and make things dynamic.... but I'm not sure we would do anything other than "wait for all endpoints to be able to show up"

jbeyerstedt, 2022-02-17:
Learned something again. Overrides would be the better solution. Thanks.

closed 2022-02-17 by jbeyerstedt