mavlink-router/mavlink-router #347: TCP Server listens only for IPv4, not IPv6

rotu, 2022-02-27:
While outgoing connections can use IPv6 addresses, `mavlink-router` does not listen for incoming IPv6 TCP connections.

E.g.
> ```
> mavlink-router version 123da4e
> Opened UDP Server [4]GCS: [::]:14550
> Opened TCP Server [5] 0.0.0.0:5760
> ```

It is suggested by the above output that the UDP server is listening on IPv6 (`[::]`) whereas the TCP server is only listening on IPv4 (`0.0.0.0`), and this indeed seems to be the case.

This creates a connection (and then immediately closes it): ```socat -4 /dev/null tcp-connect:ubuntu.local:5760```

This fails: ```socat -6 /dev/null tcp-connect:ubuntu.local:5760```

> ```
> 2022/02/26 22:46:04 socat[60098] E connect(8, LEN=28 AF=30 [2601:0444:c300:03e6:f405:0cff:fed3:f1c6]:5760, 28): Connection refused
> ```

jbeyerstedt, 2022-02-28:
The first IPv6 support in mavlink-router could be disabled all together using a build configuration switch. (I had a system with kernel IPv6 support disabled a few years back.) But this added many branches in the code, which made the code quite unreadable and was removed when switching from automake to meson.

Could you please double-check that my pull request works? I quickly connected to it with netcat, but nothing more.

rotu, 2022-02-28:
It works for me! I can connect with both IPv4 and IPv6 and exchange heartbeat messages

closed 2022-03-04 by lucasdemarchi