mavlink-router/mavlink-router #349: compilation error on Raspberry Pi Zero W 2 running 64-bit Raspbian

GMV-ROA, 2022-02-28:
[3/13] Compiling C++ object src/mavlink-routerd.p/autolog.cpp.o
FAILED: src/mavlink-routerd.p/autolog.cpp.o
c++ -Isrc/mavlink-routerd.p -Isrc -I../src -I../modules/mavlink_c_library_v2/ardupilotmega -I. -fdiagnostics-color=always -pipe -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wnon-virtual-dtor -Wextra -std=gnu++11 -O2 -g -Waddress-of-packed-member -Wno-inline -Wundef -Wformat=2 -Wlogical-op -Wsign-compare -Wformat-security -Wmissing-include-dirs -Wformat-nonliteral -Wpointer-arith -Winit-self -Wfloat-equal -Wredundant-decls -Wmissing-declarations -Wmissing-noreturn -Wshadow -Wendif-labels -Wstrict-aliasing=3 -Wwrite-strings -Wno-long-long -Wno-overlength-strings -Wno-unused-parameter -Wno-missing-field-initializers -Wno-unused-result -Wchar-subscripts -Wtype-limits -Wuninitialized -include config.h -pthread -MD -MQ src/mavlink-routerd.p/autolog.cpp.o -MF src/mavlink-routerd.p/autolog.cpp.o.d -o src/mavlink-routerd.p/autolog.cpp.o -c ../src/autolog.cpp
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.

lucasdemarchi, 2022-02-28:
out of memory probably...  ?  What if you add a swap file/partition?  It would be a better idea to crosscompile for targets that are not powerful enough to be a development platform.

GMV-ROA, 2022-02-28:
It was a memory problem, increasing the swap file size resolved it.

closed 2022-02-28 by GMV-ROA