mavlink-router/mavlink-router #350: UDP Endpoint: Add mode with fixed (broadcast) target address?

jbeyerstedt, 2022-03-04:
Currently the UDP endpoint has a quite specific functionality when using an IPv4 broadcast or IPv6 multicast address: Data will be sent as IP broadcast, but only until a packet is received which will implicitly change the target address to the remote side's address. (At least since it was fixed to it's intended functionality a few months ago.)

In some cases it might be useful to always send IP broadcast packets regardless of the incoming data. I'm not 100% sure, if this is a good idea, but it was attempted in two pull requests.
#174 was created before the IPv6 support and binds to the same address it sends data to. I think this wouldn't work with IPv6 in all cases, so PR #235 was born with two addresses: Target and listening address.

Since both PRs had no updates for a long time, I want to close them but still keep an easily findable record of the previous work, if somebody else wants to start another attempt. See https://github.com/mavlink-router/mavlink-router/pull/235#issuecomment-1039114902 for a version with less merge conflicts since quite some cleanup and restructuring work was done on mavlink-router in the meantime.

lucasdemarchi, 2022-03-04:
I'm still not sold on its usefulness. It seems to be because people want to configure just a single endpoint in mavlink-router and then have any number of systems listening and talking through that channel, instead of opening one channel/endpoint per client.

It may work on specific scenarios, but there are some potential issues:

```
                                   .----.
                                   | OS |
                                   | -- |--> sys_2
sys_1  ------>  mavlink-router --->||   |
                                   | -- |--> sys_3
                                   |    |
                                   '----'
```

- Packet from sys_2 can never reach sys_3, as it's technically the same channel, so we can't send the packet back.
- sys_2 wakes up for every packet sent from sys_1 to sys_3, just to drop it on the floor.
- It's only useful for systems on the same network, so can't even be said it adds redundancy

On the PRO side it seems to be: I want `n` GCS connected to the drone and not bother with configuring mavlink-router with n endpoints. Which says more about our lack of support for dynamic endpoint than about the need for broadcast implemented like on those PRs.  Maybe a middle ground would be for mavlink-router, when configured with a broadcast address, to spawn new endpoints once a packets are received, and allow passing only a few packets through the broadcast (only heartbeat?)

jbeyerstedt, 2022-03-07:
My implementation originated in a quick hack to broadcast (actually IPv6 multicast) just one MAVLink packet (UTM_GLOBAL_POSITION) via IEEE 802.11p's IPv6 capability for decentralized collision prevention. But MAVLink isn't meant to be used with a "public" network because sysIDs need to be unique. This concept has many issues when viewed in detail and I'm sure the implementation will create more issues than it might solve.

I submitted the PR because I thought it might be useful in other scenarios. But the only one I can come up with is broadcasting telemetry to be received by some display or data logger applications without ever receiving data from them. As stated in your comment, configuring them individually shouldn't be an insurmountable hurdle. 

We can also close this issue with "won't be implemented b/c: see above".

joaoantoniocardoso, 2022-08-04:
Hello, I'm kinda late to the party but I am feeling the pain of not having support for the following scenario, related to what is being discussed here..:

I have a system that manages several cameras as MAVLink components, which are dynamically created/deleted in runtime:
```
                                   .--------.
                                   | sys_1  |
                                   | --/\-- |
[QGC] <---> [mavlink-router] <---> | comp_1 |
                                   | comp_2 |
                                   | comp_3 |
                                   '--------'
```

In this case, `QGC` receives the heartbeats from every component of `sys_1`, but when it tries to talk with `comp_1`, it eventually succeeds but usually fails because `mavlink-router` is routing all messages to the last component it has received a heartbeat from, so sometimes `comp_3` is receiving messages sent from QGC for `comp_1` and `comp_2`,  sometimes `comp_1` is receiving it all, and the same happens with `comp_2`.

jbeyerstedt, 2022-08-04:
I think, we should discuss this in another issue.

mark-nick-o, 2022-09-01:
yes i have a similar issue i am hoping that optimising the kernel performance (overclocking) and removing the powersave options makes the applications react quick enough to receive all packets and that they are ALL received very promptly and do the various actions, i am unsure in design whether it is best to have an endpoint for each application (component action) or have one endpoint from mavlink-routerd which reads the mavlink from the FCU and then let the objects set in that receiver task pass the neccessaey information to each separate component program as their data, it must duplicate to each end-point if each component reads a seperate port and it didn't seem to getting through when i tried it under load. Ideally each output port from mavlink routerd needs to a filtered result of the traffic received in to it.