mavlink-router/mavlink-router #352: I see the 0 size log factory is still working full time

highfreq, 2022-03-13:
I see you fixed the 0 size log factory ahahahah, what a joke.

khancyr, 2022-03-13:
Before criticize, please make a correct report on how this happens.

You are using an open source projects, if you want something done, you probably need to help instead of complaining and mocking people that invest time to share a project with other for free.

And if you were really into looking for a fix for this issue, you would have read that we are discussing and trying some solutions...

highfreq, 2022-03-13:
Ma vai a fare in culo faccia di merda.

lucasdemarchi, 2022-03-13:
Interesting how you're not using it, yet you have time to waste everyone's else time. Go use your own software that you shared with everybody and market it so people know how awesome it is.

And just because you write in Italian, it doesn't mean you can use this kind of words here.

closed 2022-03-13 by lucasdemarchi