mavlink-router/mavlink-router #359: Error getting MAVLink messages from routed UDP Port through code.

darshan-k-s, 2022-04-08:
I am routing data from a UART port `/dev/ttyAMA0` to a local UDP port `127.0.0.1:6060`. I have a script `rpi_msg.py` that uses pymavlink to connect to this UDP port. However, I am not getting any data. It gets stuck on the `wait_heartbeat()` statement.
`main.conf`:
```
# This is the mavlink-router configuration file.

[General]
TcpServerPort = 0

[UartEndpoint PlannerServerPC]
Device = /dev/ttyAMA0
Baud = 57600

[UdpEndpoint LocalTransfer]
Mode = Normal
Address = 127.0.0.1
Port = 6060
```
`rpi_msg.py`:
```
from pymavlink import mavutil
import time

master = mavutil.mavlink_connection('udpin:127.0.0.1:6060')
print('UDP connection established')

master.wait_heartbeat()
print('Heartbeat received')

while True:
    try:
        print(master.recv_match().to_dict())
    except:
        pass
    time.sleep(0.1)
```

jbeyerstedt, 2022-04-08:
Could you post the output of a few seconds of `sudo mavlink-routerd -v`? Because this should work, if the "PannerServerPC" is sending heartbeats. Keep in mind, that not all ground control station software (e.g. QGroundControl) is sending heartbeats with no connected drone by default.

darshan-k-s, 2022-04-09:
This is the output :
```
mavlink-router version v2-227-gecb038b
Opened UART [4]PlannerServerPC: /dev/ttyAMA0
UART [4]PlannerServerPC: speed = 57600
Opened UDP Client [5]LocalTransfer: 127.0.0.1:6060
```

`PlannerServerPC` runs another mavlink-router, which routes data from the serial radio to another UART port, which is connected to the RPi where we route the data from the UART endpoint to a UDP port. 

Here is the config in `main.conf` of PlannerServerPC : 
```
[General]
TcpServerPort = 0

[UartEndpoint RFD]
Device = /dev/ttyUSB0
Baud = 57600

[UartEndpoint RPi]
Device = /dev/ttyUSB1
Baud = 57600
```

PlannerServerPC however opens USB1 at 115200 no matter what I set the Baud as.

jbeyerstedt, 2022-04-10:
So the chain is
```
[Serial Radio] --UART-- [PlannerServerPC] --UART-- [Raspberry Pi] --UDP-- [Python]
```
with the PlannerServerPC running the config from the second comment and the Raspberry running the first config.

If you're using an up-tp-date version (master branch) of mavlink-router on both sides, parsing the UART baud rate setting in from the config shouldn't be an issue.

My suggestion to troubleshoot this is to run mavlink-router with the verbose flag `-v` on both sides and see, if messages are received. There should be lines like `Endpoint $foo got 12 bytes` and `Endpoint $bar accepted message`, if messages are routed through mavlink-router. If there are no additional log lines, you're not receiving anything.

darshan-k-s, 2022-04-10:
Okay thanks this solved the issue. 
Now when I parse this data however, I am getting possibly corrupted messages.
```
Connection successful
Heartbeat received
{'mavpackettype': 'GLOBAL_POSITION_INT', 'time_boot_ms': 3412176, 'lat': 133474912, 'lon': 747922149, 'alt': 103734, 'relative_alt': 76, 'vx': -2, 'vy': -1, 'vz': 0, 'hdg': 34136}
{'mavpackettype': 'ATTITUDE', 'time_boot_ms': 3412221, 'roll': 0.039363645017147064, 'pitch': 0.01911311410367489, 'yaw': -0.32529446482658386, 'rollspeed': 0.0021530366502702236, 'pitchspeed': -0.001802824903279543, 'yawspeed': -0.0019392329268157482}
{'mavpackettype': 'LOCAL_POSITION_NED', 'time_boot_ms': 3412226, 'x': -2.5101163387298584, 'y': 4.66231107711792, 'z': -2.2920424938201904, 'vx': -0.022741049528121948, 'vy': -0.01080141682177782, 'vz': -0.009768708609044552}
{'mavpackettype': 'ATTITUDE', 'time_boot_ms': 3412311, 'roll': 0.03937719762325287, 'pitch': 0.019097357988357544, 'yaw': -0.3252846598625183, 'rollspeed': -0.0008815662004053593, 'pitchspeed': 0.0021151751279830933, 'yawspeed': 0.0016319649294018745}
{'mavpackettype': 'POSITION_TARGET_LOCAL_NED', 'time_boot_ms': 3412306, 'coordinate_frame': 1, 'type_mask': 63, 'x': nan, 'y': nan, 'z': nan, 'vx': nan, 'vy': nan, 'vz': nan, 'afx': 0.0, 'afy': 0.0, 'afz': 100.0, 'yaw': -0.32529017329216003, 'yaw_rate': 0.0}
{'mavpackettype': 'VFR_HUD', 'airspeed': 0.0, 'groundspeed': 0.025456232950091362, 'heading': 341, 'throttle': 0, 'alt': 103.74283599853516, 'climb': 0.010708335787057877}
{'mavpackettype': 'ATTITUDE', 'time_boot_ms': 3412411, 'roll': 0.03939783200621605, 'pitch': 0.019097965210676193, 'yaw': -0.3252735733985901, 'rollspeed': -0.00037762499414384365, 'pitchspeed': -0.0006370636983774602, 'yawspeed': 0.002211397746577859}
{'mavpackettype': 'GLOBAL_POSITION_INT', 'time_boot_ms': 3412457, 'lat': 133474910, 'lon': 747922147, 'alt': 103745, 'relative_alt': 87, 'vx': -2, 'vy': -1, 'vz': -1, 'hdg': 34136}
{'mavpackettype': 'ATTITUDE', 'time_boot_ms': 3412497, 'roll': 0.0393955335021019, 'pitch': 0.019086763262748718, 'yaw': -0.3252679109573364, 'rollspeed': -0.0006982963532209396, 'pitchspeed': 0.0013520974898710847, 'yawspeed': -0.000921224185731262}
{'mavpackettype': 'ATTITUDE', 'time_boot_ms': 3412592, 'roll': 0.03941240906715393, 'pitch': 0.019055742770433426, 'yaw': -0.3252934515476227, 'rollspeed': -0.001367755001410842, 'pitchspeed': 0.0002537097316235304, 'yawspeed': 0.002431572647765279}
{'mavpackettype': 'ATTITUDE', 'time_boot_ms': 3412682, 'roll': 0.03938711807131767, 'pitch': 0.019017482176423073, 'yaw': -0.32532280683517456, 'rollspeed': 0.000877544516697526, 'pitchspeed': 0.0013337904820218682, 'yawspeed': -0.0020817085169255733}
{'mavpackettype': 'ATTITUDE_TARGET', 'time_boot_ms': 3412708, 'type_mask': 0, 'q': [0.9868013262748718, -0.0, 0.0, -0.16193559765815735], 'body_roll_rate': -0.2560410499572754, 'body_pitch_rate': -0.1235843375325203, 'body_yaw_rate': 0.001104030990973115, 'thrust': 0.0010000000474974513}
{'mavpackettype': 'VFR_HUD', 'airspeed': 0.0, 'groundspeed': 0.023516518995165825, 'heading': 341, 'throttle': 0, 'alt': 103.75065612792969, 'climb': 0.011097706854343414}
{'mavpackettype': 'GLOBAL_POSITION_INT', 'time_boot_ms': 3412738, 'lat': 133474908, 'lon': 747922146, 'alt': 103751, 'relative_alt': 93, 'vx': -2, 'vy': -1, 'vz': -1, 'hdg': 34135}
{'mavpackettype': 'ATTITUDE', 'time_boot_ms': 3412783, 'roll': 0.03940901532769203, 'pitch': 0.018997009843587875, 'yaw': -0.32532989978790283, 'rollspeed': 0.00011510448530316353, 'pitchspeed': 0.0018146942602470517, 'yawspeed': -8.470495231449604e-06}
{'mavpackettype': 'GPS_RAW_INT', 'time_usec': 3412649447, 'fix_type': 4, 'lat': 133474885, 'lon': 747922132, 'alt': 86638, 'eph': 68, 'epv': 97, 'vel': 2, 'cog': 0, 'satellites_visible': 17}
{'mavpackettype': 'BAD_DATA', 'data': array('B', [254, 32, 43, 1, 1, 141, 36, 153, 107, 203, 0, 0, 0, 0, 99, 216, 4, 67, 188, 129, 207, 66, 168, 187, 19, 64, 192, 100, 195, 61, 103, 64, 32, 136, 214, 61, 202, 241, 254, 2]), 'reason': 'invalid MAVLink CRC in msgID 141 0x02fe should be 0x8b1b'}
{'mavpackettype': 'BAD_DATA', 'data': bytearray(b','), 'reason': 'Bad prefix'}
{'mavpackettype': 'BAD_DATA', 'data': bytearray(b'\x01'), 'reason': 'Bad prefix'}
{'mavpackettype': 'BAD_DATA', 'data': bytearray(b'\x01'), 'reason': 'Bad prefix'}
{'mavpackettype': 'BAD_DATA', 'data': bytearray(b'\xf5'), 'reason': 'Bad prefix'}
{'mavpackettype': 'BAD_DATA', 'data': bytearray(b'\x00'), 'reason': 'Bad prefix'}
{'mavpackettype': 'BAD_DATA', 'data': bytearray(b'\x01'), 'reason': 'Bad prefix'}
{'mavpackettype': 'BAD_DATA', 'data': bytearray(b'+'), 'reason': 'Bad prefix'}
{'mavpackettype': 'BAD_DATA', 'data': bytearray(b'\n'), 'reason': 'Bad prefix'}
```
I tried lowering the baud rate to 9600 for `[PlannerServerPC] --UART-- [Raspberry Pi]` but the issue doesnt seem to be resolved.

lucasdemarchi, 2022-04-10:
If the issue was with the UART transmission, mavlink-router would rather drop the message. Since you are receiving the packet, it probably means it's something else. Do you have a matching mavlink library between the flight stack, mavlink-router and python scripts?

closed 2022-04-17 by darshan-k-s