mavlink-router/mavlink-router #364: sudo ninja -C build install ERROR

matbord, 2022-06-23:
I'm following your tutorial to install Mavlink router and when I execute "sudo ninja -C build install" I obtain this error (repeated for [2/5] [3/5] :
`ninja: Entering directory `build'
[2/5] Compiling C++ object src/mavlink-routerd.p/logendpoint.cpp.o
FAILED: src/mavlink-routerd.p/logendpoint.cpp.o 
c++ -Isrc/mavlink-routerd.p -Isrc -I../src -I../modules/mavlink_c_library_v2/ardupilotmega -I. -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wnon-virtual-dtor -Wextra -std=gnu++11 -O2 -g -Wno-inline -Wundef -Wformat=2 -Wlogical-op -Wsign-compare -Wformat-security -Wmissing-include-dirs -Wformat-nonliteral -Wpointer-arith -Winit-self -Wfloat-equal -Wredundant-decls -Wmissing-declarations -Wmissing-noreturn -Wshadow -Wendif-labels -Wstrict-aliasing=3 -Wwrite-strings -Wno-long-long -Wno-overlength-strings -Wno-unused-parameter -Wno-missing-field-initializers -Wno-unused-result -Wchar-subscripts -Wtype-limits -Wuninitialized -include config.h -pthread -MD -MQ src/mavlink-routerd.p/logendpoint.cpp.o -MF src/mavlink-routerd.p/logendpoint.cpp.o.d -o src/mavlink-routerd.p/logendpoint.cpp.o -c ../src/logendpoint.cpp
In file included from ../src/logendpoint.h:20:0,
                 from ../src/logendpoint.cpp:18:
../src/common/conf_file.h:156:25: error: expected primary-expression before ‘,’ token
         offsetof(_struct, _field), sizeof(_struct::_field) \
                         ^
../src/logendpoint.cpp:50:69: note: in expansion of macro ‘OPTIONS_TABLE_STRUCT_FIELD’
     {"Log",             false, ConfFile::parse_stdstring,           OPTIONS_TABLE_STRUCT_FIELD(LogOptions, logs_dir)},
                                                                     ^
../src/logendpoint.cpp:50:108: error: ‘logs_dir’ was not declared in this scope
     {"Log",             false, ConfFile::parse_stdstring,           OPTIONS_TABLE_STRUCT_FIELD(LogOptions, logs_dir)},
                                                                                                            ^
../src/common/conf_file.h:156:27: note: in definition of macro ‘OPTIONS_TABLE_STRUCT_FIELD’
         offsetof(_struct, _field), sizeof(_struct::_field) \
                           ^
../src/common/conf_file.h:156:33: error: ‘offsetof’ was not declared in this scope
         offsetof(_struct, _field), sizeof(_struct::_field) \
                                 ^
../src/logendpoint.cpp:50:69: note: in expansion of macro ‘OPTIONS_TABLE_STRUCT_FIELD’
     {"Log",             false, ConfFile::parse_stdstring,           OPTIONS_TABLE_STRUCT_FIELD(LogOptions, logs_dir)},
                                                                     ^
../src/common/conf_file.h:156:25: error: expected primary-expression before ‘,’ token
         offsetof(_struct, _field), sizeof(_struct::_field) \
                         ^
../src/logendpoint.cpp:51:69: note: in expansion of macro ‘OPTIONS_TABLE_STRUCT_FIELD’
     {"LogMode",         false, LogEndpoint::parse_log_mode,         OPTIONS_TABLE_STRUCT_FIELD(LogOptions, log_mode)},
                                                                     ^
../src/logendpoint.cpp:51:108: error: ‘log_mode’ was not declared in this scope
     {"LogMode",         false, LogEndpoint::parse_log_mode,         OPTIONS_TABLE_STRUCT_FIELD(LogOptions, log_mode)},
                                                                                                            ^
../src/common/conf_file.h:156:27: note: in definition of macro ‘OPTIONS_TABLE_STRUCT_FIELD’
         offsetof(_struct, _field), sizeof(_struct::_field) \
                           ^
../src/common/conf_file.h:156:33: error: ‘offsetof’ was not declared in this scope
         offsetof(_struct, _field), sizeof(_struct::_field) \
                                 ^
../src/logendpoint.cpp:51:69: note: in expansion of macro ‘OPTIONS_TABLE_STRUCT_FIELD’
     {"LogMode",         false, LogEndpoint::parse_log_mode,         OPTIONS_TABLE_STRUCT_FIELD(LogOptions, log_mode)},
                                                                     ^
../src/common/conf_file.h:156:25: error: expected primary-expression before ‘,’ token
         offsetof(_struct, _field), sizeof(_struct::_field) \
                         ^
../src/logendpoint.cpp:52:69: note: in expansion of macro ‘OPTIONS_TABLE_STRUCT_FIELD’
     {"MavlinkDialect",  false, LogEndpoint::parse_mavlink_dialect,  OPTIONS_TABLE_STRUCT_FIELD(LogOptions, mavlink_dialect)},
                                                                     ^
../src/logendpoint.cpp:52:108: error: ‘mavlink_dialect’ was not declared in this scope
     {"MavlinkDialect",  false, LogEndpoint::parse_mavlink_dialect,  OPTIONS_TABLE_STRUCT_FIELD(LogOptions, mavlink_dialect)},
                                                                                                            ^
../src/common/conf_file.h:156:27: note: in definition of macro ‘OPTIONS_TABLE_STRUCT_FIELD’
         offsetof(_struct, _field), sizeof(_struct::_field) \
                           ^
../src/common/conf_file.h:156:33: error: ‘offsetof’ was not declared in this scope
         offsetof(_struct, _field), sizeof(_struct::_field) \
                                 ^
../src/logendpoint.cpp:52:69: note: in expansion of macro ‘OPTIONS_TABLE_STRUCT_FIELD’
     {"MavlinkDialect",  false, LogEndpoint::parse_mavlink_dialect,  OPTIONS_TABLE_STRUCT_FIELD(LogOptions, mavlink_dialect)},
                                                                     ^
../src/common/conf_file.h:156:25: error: expected primary-expression before ‘,’ token
         offsetof(_struct, _field), sizeof(_struct::_field) \
                         ^
../src/logendpoint.cpp:53:69: note: in expansion of macro ‘OPTIONS_TABLE_STRUCT_FIELD’
     {"MinFreeSpace",    false, ConfFile::parse_ul,                  OPTIONS_TABLE_STRUCT_FIELD(LogOptions, min_free_space)},
                                                                     ^
../src/logendpoint.cpp:53:108: error: ‘min_free_space’ was not declared in this scope
     {"MinFreeSpace",    false, ConfFile::parse_ul,                  OPTIONS_TABLE_STRUCT_FIELD(LogOptions, min_free_space)},
                                                                                                            ^
[... 22 lines cut ...]
         offsetof(_struct, _field), sizeof(_struct::_field) \
                                 ^
../src/logendpoint.cpp:54:69: note: in expansion of macro ‘OPTIONS_TABLE_STRUCT_FIELD’
     {"MaxLogFiles",     false, ConfFile::parse_ul,                  OPTIONS_TABLE_STRUCT_FIELD(LogOptions, max_log_files)},
                                                                     ^
../src/common/conf_file.h:156:25: error: expected primary-expression before ‘,’ token
         offsetof(_struct, _field), sizeof(_struct::_field) \
                         ^
../src/logendpoint.cpp:55:69: note: in expansion of macro ‘OPTIONS_TABLE_STRUCT_FIELD’
     {"LogSystemId",     false, LogEndpoint::parse_fcu_id,           OPTIONS_TABLE_STRUCT_FIELD(LogOptions, fcu_id)},
                                                                     ^
../src/logendpoint.cpp:55:108: error: ‘fcu_id’ was not declared in this scope
     {"LogSystemId",     false, LogEndpoint::parse_fcu_id,           OPTIONS_TABLE_STRUCT_FIELD(LogOptions, fcu_id)},
                                                                                                            ^
../src/common/conf_file.h:156:27: note: in definition of macro ‘OPTIONS_TABLE_STRUCT_FIELD’
         offsetof(_struct, _field), sizeof(_struct::_field) \
                           ^
../src/common/conf_file.h:156:33: error: ‘offsetof’ was not declared in this scope
         offsetof(_struct, _field), sizeof(_struct::_field) \
                                 ^
../src/logendpoint.cpp:55:69: note: in expansion of macro ‘OPTIONS_TABLE_STRUCT_FIELD’
     {"LogSystemId",     false, LogEndpoint::parse_fcu_id,           OPTIONS_TABLE_STRUCT_FIELD(LogOptions, fcu_id)},
`

Any idea to help me to solve it, please?

lucasdemarchi, 2022-06-26:
What compiler and compiler version are you using? Also always make sure you build with your normal user (without sudo). Only use sudo to install on system directory.

lucasdemarchi, 2022-07-20:
closing due to inactivity, please reopen if issue persists

closed 2022-07-20 by lucasdemarchi