mavlink-router/mavlink-router #365: crash of mavlink-routerd

mark-nick-o, 2022-06-29:
after this running for sometime without issue (we have increased the frequency of messages to 10Hz from 1Hz from the FCU)

mavlink-routerd is crashing with the following errors when i try to run it
pi@CameraPi:~ $ /usr/bin/cpulimit --limit=30 /usr/bin/mavlink-routerd  -e 10.0.2.51:14550 -e 127.0.0.1:14551 /dev/ttyACM0:115200
mavlink-router version v2-146-g9261530
Could not open /dev/ttyACM0 (Device or resource busy)

i upgraded it to the latest version here as the one i was running was 88 comits behind but i also still get this on re-spawn (Its being cpulimited on start-up to 30% for load, and this did work okay last week)
pi@CameraPi:~ $ /usr/bin/cpulimit --limit=30 /usr/bin/mavlink-routerd  -e 10.0.2.51:14550 -e 127.0.0.1:14551 /dev/ttyACM0:115200
mavlink-router version v2-229-g022333d
Could not open /dev/ttyACM0 (Device or resource busy)

but the device is there and its the port the the flight control unit FCU it should be closed in the handler 
pi@CameraPi:~ $ ls /dev/ttyACM1
/dev/ttyACM1

Secondly i saw this error also when the mavlink-routerd crashed one time.
TCP Address already in use
i cant understand that as your code already has this or something similar which should allow the reuse when in TIME_WAIT state
std::bool yes = 1;
setsockopt(sock, SOL_SOCKET, SO_REUSEADDR, (const char *)&yes, sizeof(yes));
and thats in the tcp.open method ???

and thidly i got this error
poll error for fd 4
Critical fd 4 got error, exiting
I dont know what this means ????

Is anyone able to explain this strange behaviour I'm seeing and point me towards a resolution

mark-nick-o, 2022-06-29:
this is the other error im getting as above

mavlink-router version v2-229-g022333d
Opened UART [4]CLI: /dev/ttyACM0
UART [4]CLI: speed = 115200
Opened UDP Client [5]CLI: 10.0.2.51:14550
Opened UDP Client [7]CLI: 127.0.0.1:14551
TCP Server: Could not bind to tcp socket (Address already in use)

Process 14384 terminated abnormally

and with setsocket the serial port problem shows used except there is no more mavlink-routerd running and no other application opening this port, so i cant see why it gives me the above errors some of the time.

pi@CameraPi:~ $ setserial -g /dev/ttyACM[0123]
/dev/ttyACM0: Device or resource busy
/dev/ttyACM1, UART: unknown, Port: 0x0000, IRQ: 0

the effect is that the program reading the UDP endpoint is also going down all the time with this ??

mark-nick-o, 2022-06-29:
after reboot some of these errors cleared and mavlink-routerd then started up nicely
however on connecting to the end point it dies like this

poll error for fd 4
Critical fd 4 got error, exiting

lucasdemarchi, 2022-06-29:
```
poll error for fd 4
Critical fd 4 got error, exiting
```

fd 4, per your logs above correspond to:

```
Opened UART [4]CLI: /dev/ttyACM0
UART [4]CLI: speed = 115200
```

It means that for some reason it failed to open the UART.  Is that serial port working reliably? Are you able to open `screen /dev/ttyACM0 115200` and see the data coming in?

What board is this?

lucasdemarchi, 2022-06-29:
> but the device is there and its the port the the flight control unit FCU it should be closed in the handler
> pi@CameraPi:~ $ ls /dev/ttyACM1
> /dev/ttyACM1

But it complains that the device that failed is /dev/ttyACM0, not /dev/ttyACM1

mark-nick-o, 2022-06-29:
making sure the code reading off the endpoint first before starting the mavlink-routerd seemed to work for me

lucasdemarchi, 2022-06-29:
if the port vanished, it may have been disconnected (if it's usb). But per the name it seems to be the SoC uart, so it's likely a kernel driver bug. You will have to look at `dmesg` to check if kernel says anything when that happens.

lucasdemarchi, 2022-06-29:
and make sure you have an up-to-date kernel

lucasdemarchi, 2022-06-29:
> making sure the code reading off the endpoint first before starting the mavlink-routerd seemed to work for me

I think this is coincidence. If it's a kernel  driver bug, then it will work when you boot and later stop working.

mark-nick-o, 2022-06-29:
thanks very much ill check it when it re-occurs it shows fine right now again, i think ill look for the up to date kernel its possible something else i saw unrelated also may be the kernel

mark-nick-o, 2022-06-29:
kernel did need update and its in progress
for info to others its :-
sudo apt update
sudo apt full-upgrade
sudo rpi-update

lucasdemarchi, 2022-07-20:
did you have any issues after doing the updates?

We should probably change mavlink-router to avoid crashing, but the fix seems to be on the kernel side.

mark-nick-o, 2022-07-21:
no issue it upgraded fine and seemed to be stable, that seemed to fix the issue. Using a tcp endpoint seemed more stable than the udp but it might just have been co-incidence, just left that method for now, and i didnt see the fault again yet.

lucasdemarchi, 2022-07-21:
Thanks for confirming. Closing the issue.

closed 2022-07-21 by lucasdemarchi