mavlink-router/mavlink-router #367: Can this project be used on raspberry pi ?

Metear, 2022-07-15:


jbeyerstedt, 2022-07-15:
Yes. Just use the build instructions for Debian/Ubuntu when you're using Raspbian

Metear, 2022-07-15:
Header <aio.h> has symbol "aio_init" : YES 
Configuring config.h using configuration

meson.build:127:0: ERROR: Include dir modules/mavlink_c_library_v2/ardupilotmega does not exist.

I have this err when run the command (ninja -C build && sudo ninja -C build install) .
My configuration environment is Raspberry Pi 3B+ and ubuntu20 server.

Metear, 2022-07-15:
Sorry, probably due to my network issues, I have resolved this after re-running git submodule update --init --recursive.

thealienthing, 2022-07-19:
I would recommend checking out the Dockerfiles in the examples directory. It will compile a working container for you that can run on your system architecture. I've built mavlink-router with the dockerfile many times on the Nvidia Jetson Xavier which uses an arm64 architecture. It should be no different on raspberry pi and should be very easy to set up.

lucasdemarchi, 2022-07-20:
Also, it's a good idea to build it on your machine via a container or even with pi-gen as @booo  is using. This avoids situations like trying to build on rpi 0 and running out of memory.

closed 2022-07-20 by lucasdemarchi