mavlink-router/mavlink-router #368: Filtering by SrcSysID does not function as intended (bug)

max-itsl, 2022-07-26:
**TLDR: AllowSrcSysIn filter does not work at all, but AllowSrcSysOut works as a two-way whitelist filter (IN and OUT).**

**Test setup scheme:**

[image]

Firmware: latest PX4 1.13 stable
Hardware: Cube Black, Cube Orange, Holybro Pixhawk 4, Matek H743 Slim
GCS: Recent QGroundControl Daily build

### Test case 1: No filtering applied

Configuration file content /etc/mavlink-router/main.conf  :
```
[General]
TcpServerPort = 5760
[UartEndpoint ToRover15]
Device = /dev/ttyTHS1
Baud = 57600
```

**Test 1 result:** Everything works fine. Can easily connect from both instances of QGroundControl (SysID 233 and 241 at the same time). Parameters download procedure is very fast. Can upload and download missions.

Debug console output: sudo mavlink-routerd -g debug
```
mavlink-router version v2-229-g022333d
Opened UART [4]ToRover15: /dev/ttyTHS1
UART [4]ToRover15: speed = 57600
Opened TCP Server [5] [::]:5760
…
…
TCP Server: New client
TCP [7]dynamic: Connection accepted
…
…
> UART [4]ToRover15: Got 40 bytes
Endpoint [4]ToRover15: got message 30 to -1/-1 from 15/1
	Known components:
		15/1
Endpoint [7]dynamic: got message 30 to -1/-1 from 15/1
	Known components:
Endpoint [7] accepted message 30 to -1/-1 from 15/1
TCP [7]dynamic: Wrote 40 bytes
…
…
> TCP [7]dynamic: Got 21 bytes
Endpoint [4]ToRover15: got message 0 to -1/-1 from 233/190
	Known components:
		15/1
Endpoint [4] accepted message 0 to -1/-1 from 233/190
UART [4]ToRover15: Wrote 21 bytes
Endpoint [7]dynamic: got message 0 to -1/-1 from 233/190
	Known components:
		233/190
…
…
TCP Server: New client
TCP [8]dynamic: Connection accepted
> UART [4]ToRover15: Got 29 bytes
Endpoint [4]ToRover15: got message 74 to -1/-1 from 15/1
	Known components:
		15/1
Endpoint [7]dynamic: got message 74 to -1/-1 from 15/1
	Known components:
		233/190
Endpoint [7] accepted message 74 to -1/-1 from 15/1
TCP [7]dynamic: Wrote 29 bytes
Endpoint [8]dynamic: got message 74 to -1/-1 from 15/1
	Known components:
Endpoint [8] accepted message 74 to -1/-1 from 15/1
[... 286 lines cut ...]
Endpoint [8] accepted message 33 to -1/-1 from 15/1
TCP [8]dynamic: Wrote 40 bytes
…
…

```

### Test case 6: add all SysIDs
**OK, lets add all SysID to allowed and see what happens.**

Configuration file content /etc/mavlink-router/main.conf :
```
[General]
TcpServerPort = 5760
[UartEndpoint ToRover15]
Device = /dev/ttyTHS1
Baud = 57600
AllowSrcSysOut = 15,233,241
```

**Test 6 result: Everything works fine.** Can easily connect from both instances of QGroundControl (SysID 233 and 241 at the same time). Parameters download procedure is very fast. Can upload and download missions.

**My initial assumption is once again proved – AllowSrcSysOut acts as a two-way whitelist filter.**

**P.S.** Also tried various scenarios using several rovers and intermediate mavlink-routers, various combinations of endpoints TCP clients, UDP servers and clients, UART ports – all points out that AllowSrcSysIn filter does not work, but AllowSrcSysOut works as a two-way whitelist filter.

jbeyerstedt, 2022-08-01:
Thanks for the detailed analysis. It was a simple copy-paste error when passing the config to the endpoint configuration. Sorry for that.

closed 2022-08-01 by lucasdemarchi

max-itsl, 2022-08-01:
Good day! Thank you! Will test today.

Regards
Max

PR #369 (2022-08-01): Filtering: Fix AllowSrcSysIn
Copy-paste error when initializing the input filtering from the config structs.

closes #368